CalgaryToSpace/CTS-SAT-1-OBC-Firmware #13: Integrate ADCS system functions, when complete

DeflateAwning, 2024-03-10:
ADCS functions (I2C interface) are being developed separately. Integrate them into this repo when complete.

DeflateAwning, 2024-05-25:
Chatted with @DeemDeem52 on the path forward with this repo and using telecommands! He's working on it, and will get a unit test done for Tuesday so we can discuss any issues had!

DeemDeem52, 2024-06-01:
I've gotten started updating the functions for unit testing and telecommands. It's going to take some time, though. 
(The ADCS team is having problems right now, so we're not actually *losing* any time, though)

DeemDeem52, 2024-06-08:
Refactoring is making excellent progress. I've got unit tests up and running, which are showing me that I need to do some bugfixing. The ADCS team has also maganed to fix the problem they were having, so testing can also resume!

DeflateAwning, 2024-06-08:
Fantastic news and updates! Happy to hear that the unit tests are helping fix bugs before they're hard to detect later on; that's the ideal scenario!

DeemDeem52, 2024-06-25:
Refactoring of the existing functions is just about complete (barring some minor things which don't affect how the code runs -- those will be fixed after I've got all the code up and functional)! Additional news, the latest bugfix combined with some troubleshooting from the ADCS team has fixed the issues with the ADCS, so we're back up and running for testing again. Although for some reason I seem to have created two divergent branches in my attempts to rename the ADCSFirmware branch... which somehow still has the latest commits. I'll get on fixing that once I can figure out how it happened.

DeflateAwning, 2024-07-06:
How's the progress of the testing? I'd love to see testing and finalize the structure of just a couple telecommands to minimize any reworking we have to do across _all_ the telecommands if there are issues/opportunities for improvement.

DeemDeem52, 2024-07-06:
Our next testing date will be Saturday the 20th, so we'll likely be testing the telecommands then.

DeemDeem52, 2024-07-27:
Continuing along with this.....
Implementing the CTS1 telecommands is proving slightly more difficult than anticipated -- once my other pull request is merged I'll do a rebase (which I'll need to do anyways but will probably be painful) and then should be able to continue with this.

In other news, all of the packer function unit tests are complete and the functions pass the test! I caught a good few bugs in that, so it's really good that those are done.

DeemDeem52, 2024-07-27:
So, as a whole, my to-do list for this looks like:
- Write and test the CTS1 telecommands
- Integrate the functions I've written with the memory module somehow (need to consult someone on this)
- Fix up any comments or docstrings that aren't quite right
- Minor function changes: ADCS_Init() for the CRC8, any other functions that are needed or wanted
- Cleanup: removing the deprecated UART functions and duplicate I2C scanner function, etc

DeflateAwning, 2024-08-02:
Can we please focus on implement 2 ADCS commands, with 2 associated telecommands, and all the other functions required to make this work? You can pick any 2 commands/functions. Let's ignore the rest until we get that merged into `main`.

We need to get _something_ stabilized so we can divide up the rest of the work, or re-evaluate our base driver strategy.

Sorry the initial Issue was scoped so broadly; this will be better.

DeemDeem52, 2024-08-02:
Well, that's definitely my first test case here. But I don't think you need to worry about the broad scope; once I have the first couple telecommands working, the rest (which is already nearly done) will be a piece of cake!

DeemDeem52, 2024-08-05:
I've written something for all of the telecommands in the commissioning manual, and cleaned up a good chunk of the warnings. I won't be able to make the telecommands work until I next test with the ADCS test device in the cleanroom, so my next objective is to see what I need to do to integrate with the memory module / data downlink protocols for telemetry requests.

DeflateAwning, 2024-08-05:
Can you first please create a really good PR with only about 2 ADCS commands, with 2 associated telecommands, and all the other functions required to make this work? You can pick any 2 commands/functions. And then can you please request a review on it so we can check through if there's any major opportunities for improvement with what you have so far?

DeemDeem52, 2024-08-05:
I think I can do that, though I'm not sure how to split the functions via GitHub from the current branch so as to create a PR.

closed 2024-09-21 by DeemDeem52