CalgaryToSpace/CTS-SAT-1-OBC-Firmware #14: Create "MPI active and recording data" mode

DeflateAwning, 2024-03-10:
### Telecommands:
1. Turn MPI on and start recording
     * Arg: Run duration in seconds, from 1 sec to 20 minutes.
2. Turn off MPI early
     * Turns the MPI off immediately, if it's active currently

### Requirements, when active and recording
* Start a file with the "recording start time" timestamp.
* Collect the data and write it to the filesystem.
* Monitor temperature (from EPS), and stop experiment early if too hot
* Monitor temperature (from MPI), and stop experiment early if too hot

DeflateAwning, 2024-09-13:
@JohnathanBurchill, in the current plan, we want to make MPI data recording very "high priority" in scheduling sense, but we don't want it to cause essential satellite functionality to stop working. For example, receiving telecommands, logging GPS data, etc. must continue during MPI data collection.

As such, it is possible that, perhaps, every few seconds, a few frames of MPI data will be dropped (in a predictable/expected/documented manner) to ensure satellite operation continues smoothly. Obviously the goal is to use features like Direct Memory Access (DMA), thread/task management, etc. to minimize the dropped data.

Can you give some suggestions on allowable vs. problematic thresholds for this collection? Is dropping the following amount of data reasonable?

* 1 full contiguous second per 30-second period of data collection
* 2 frames every 3 seconds

Note that each frame will be timestamped, so dropped frames will not "skew" the data, but will instead just appear as gaps (if analyzed appropriately).

JohnathanBurchill, 2024-09-13:
> @JohnathanBurchill, in the current plan, we want to make MPI data recording very "high priority" in a scheduling sense, but we don't want it to cause essential satellite functionality to stop working. For example, receiving telecommands, logging GPS data (if enabled), etc. must continue during MPI data collection.
> 
> As such, it is possible that, perhaps, every few seconds, a few frames of MPI data will be dropped (in a predictable/expected/documented manner) to ensure satellite operation continues smoothly. Obviously the goal is to use features like Direct Memory Access (DMA), thread/task management, etc. to minimize the dropped data.
> 
> Can you give some suggestions on allowable vs. problematic thresholds for this collection? Is dropping the following amount of data reasonable?
> 
> * 1 full contiguous second per 30-second period of data collection
> * 2 frames every 3 seconds
> 
> Note that each frame will be timestamped, so dropped frames will not "skew" the data, but will instead just appear as gaps (if analyzed appropriately).

As a starting point, either way is fine. I'd like to aim to get all the data though. Happy to discuss.

The MPI bytes do not stream continuously. At a baud rate of 23400, the 150 bytes of an MPI frame take about 6 ms to transmit to the OBC. The frame rate is about 32 per second - much of the MPI operation involves integrating ion signal (31 ms), and that is typically a lot longer than the time it takes to transmit a frame (6 ms). Maybe there is a way to build the byte capture into the UART interrupt using a ram buffer with flags to signal  a lower-priority task to write the buffer to a drive.

DeflateAwning, 2024-09-13:
That sounds much more manageable than I had in my head. 150 bytes/frame x 32 frames/s = 4800 bytes/sec.

If we use a buffer of 19,200 bytes (4800x4), the writeback (which occurs when the buffer is half-full) could happen once every 2 seconds. This will be easy to hit.

I don't think we'll need to drop any data anymore.

2BlackCats, 2025-02-28:
Hey, @vaibhavkpr are you still working on this issue?

Saksham-P, 2025-05-31:
Yea, we hope to return the ISR as quickly as possible, so if the proposed implementation does end up being too slow, we could shift the temperature check to the "buffer to File System" thread.

For the threshold, is it something that could potentially be changed during the mission? If so, I could create a separate telecommand to change the thresh.

JohnathanBurchill, 2025-05-31:
> For the threshold, is it something that could potentially be changed during the mission? If so, I could create a separate telecommand to change the thresh.

It may change during the mission. You would not need a separate telecommand: I think telecommands are already implemented for reporting and updating system configuration variables.

DeflateAwning, 2025-05-31:
Let's move discussion about the temperature threshold to this new issue: #462 

Let's get this issue solved and merged first using a smaller DMA buffer so that timestamps can be inlayed with the core MPI data. I believe @Saksham-P should have most of a solution for that solved now.

PR #486 (2025-06-02): MPI science data collection via DMA
Superseded #463. Full credit to @Saksham-P for the bulk of this implementation.

Resolves #14.

Related to #464, but might not be a complete solution for it.
closed 2025-06-02 by DeflateAwning