CalgaryToSpace/CTS-SAT-1-OBC-Firmware #16: GPS: log GPS position to filesystem

DeflateAwning, 2024-03-10:
* Write a function which logs the date and time to the GPS log file in the filesystem. 
    * Function argument: how many seconds of logs to collect (default: 60)
* Cycle the log every 24h to a new file.
* Call this function once-per-hour (interval configurable with a configuration variable).

#### Notes
* GPS requires a "warm-up" period before recording data.

mzcigler14, 2024-06-27:
added GPS files to the new main (the one using VSCode.

mzcigler14, 2024-06-27:
Added GPS_init file with initialize_GPS command to test GPS without affecting main.c

mzcigler14, 2024-06-27:
added all commands to GPS_types

DeflateAwning, 2024-06-27:
Would love to see this work! Would you mind pushing a branch named something like: `matt/gps-logging-i16`

mzcigler14, 2024-07-03:
there is a branch currently matjaz-GPS, let me know If you can see it

mzcigler14, 2024-07-03:
finished code that parses the receive array if there are multiple logs in one receive, started writing out test prints, need to complete this and test with rasberry pi to make sure parsing is working properly.

DeflateAwning, 2024-08-17:
Question for the Orbit team to answer here:

1. How do we know when the GPS "has a fix", and that its timing and position data is valid, based on the response to commands?
2. Are there GPS "states" when the time it provides is valid, but the positions is not valid?
3. Is there a way to get the uncertainty of the timestamp?
4. How does the PPS (pulse-per-second) output pin work? Does it pulse at the "top" of each second (i.e., when the number of milliseconds is 0)? Is any configuration required?

DeflateAwning, 2024-08-30:
Two commands to focus on parsing, only. The rest will be handled by a future mechanism in #138 (arbitrary GPS commands).

Two main commands:
* BESTXYZ
* TIME

In BESTXYZ, parse out, and later make very easy to log/downlink the following fields:
* pos status
* pos type
* Location coordinates
* Location standard deviation
* diff_age
* sol_age

In TIME, parse out into a struct, and later use to update the clock:
* clock status (maybe not needed)
* utc offset
* utc status (valid, invalid, warning)

christopherkinyua, 2024-10-15:
> @christopherkinyua was wondering what the status on this Issue is?

Currently conducting tests with the GPS receiver. I need to solder pin headers onto my STM board then I can carry on with the testing.

DeflateAwning, 2025-05-12:
Closing this as completed due to the awesome `@resp_fname` telecommand suffix tag.

closed 2025-05-12 by DeflateAwning