CalgaryToSpace/CTS-SAT-1-OBC-Firmware #17: Configuration variables: create a data structure which stores all configuration variables, and can downlink them all

DeflateAwning, 2024-03-10:
### Telecommand: Set configuration values
Not sure if it's best to have separate commands for each variable, or if it's best to have a command with two args ("which variable", and "new value").

### Telecommand: Fetch specific configuration value
Fetches one variable's value.
**Arg:** which variable to fetch

### Telecommand: Fetch all configuration values
...

NuclearTea, 2024-06-10:
Took me a while to create a function which could split a string by a delimiter though it is done for the most part. Left comments describing function and possible improvements.

Now need to discuss the structure of configuration vars with Parker to ensure that I am doing it right

DeflateAwning, 2024-06-10:
Can you push a branch to take a look at? 

Do you have a specific question? Otherwise, can you propose a solution and we can make revisions from there?

NuclearTea, 2024-06-10:
> ?

I should have linked it with this issue. It has the word ConfigurationVariable in it

DeflateAwning, 2024-06-10:
> It has the word ConfigurationVariable in it.

Don't see it, sorry.

I think we can simplify that even further and just include the subsystem name in the configuration variable name. Sounds reasonable otherwise, though!

DeflateAwning, 2024-06-10:
Feedback on the progress:

1. Avoid reformatting existing code in an unrelated branch/PR (e.g., `firmware/Core/Src/unit_tests/unit_test_inventory.c`). If you'd like to setup an autoformatter or similar, then take a look at the applicable issue. 
2. Please rebase instead of merging "main" into your feature branch.
3. How are you going to handle different configuration variable types (e.g., uint8_t, uint16_t, etc.)? A void pointer doesn't know what data type it points to.
4. Please name functions like `TEST_EXEC__HELPER_split_string_by_delimiter()` instead of `TEST_EXEC__HELPER_Split_String_By_Delimiter()`. Also, name them like `fruit_output` instead of `fruitOutput`.
5. By convention, please use `char` datatype for null-terminated strings, and `uint8_t` for bytes/byte arrays.
6. Avoid extremely generic names like "helpers.h". What is it helping? Is it an array function? Is it for strings? So far, the `transforms/` folder has some of these general functions, so perhaps add to there.

NuclearTea, 2024-06-11:
> Looks good overall!
> 
> Feedback on the progress:
> 
> 1. Avoid reformatting existing code in an unrelated branch/PR (e.g., `firmware/Core/Src/unit_tests/unit_test_inventory.c`). If you'd like to setup an autoformatter or similar, then take a look at the applicable issue.
> 2. Please rebase instead of merging "main" into your feature branch.
> 3. How are you going to handle different configuration variable types (e.g., uint8_t, uint16_t, etc.)? A void pointer doesn't know what data type it points to.
> 4. Please name functions like `TEST_EXEC__HELPER_split_string_by_delimiter()` instead of `TEST_EXEC__HELPER_Split_String_By_Delimiter()`. Also, name them like `fruit_output` instead of `fruitOutput`.
> 5. By convention, please use `char` datatype for null-terminated strings, and `uint8_t` for bytes/byte arrays.
> 6. Avoid extremely generic names like "helpers.h". What is it helping? Is it an array function? Is it for strings? So far, the `transforms/` folder has some of these general functions, so perhaps add to there.

1. I am using the C/C++ VS Code Extension which comes with a default formatter, I would assume that everyone is using the same, so wouldn't be tough to integrate
2. ✅ 
3. I want to do this generically so that its less code and less branching (which I have found can become difficult to debug once its' 3+ levels deep). I could add another property which specifies the type (though I don't like this approach). We could also leave it to the users discretion (not a great solution either). I really don't want to manually set each variable in a huge if-elif statement lol.
4. ✅ 
5. ✅ 
6. ✅ Renamed to string_helpers every where

NuclearTea, 2024-06-11:
Another solution that I can think of is the following:

2D Array:

uin8_t Configuration_Variable_Table[][] = { EPS_Configuration_Vars, ADCS_Configuration_Vars, ...}
Rows are the Subsystem enum values. Ex: 

```
typedef enum {
CTS_Subsystem_EPS,
CTS_Subsystem_ADCS,
...
} CTS_Subsystem
```
So you could get EPS_Configuration_Variables by doing the following:

```
Configuration_Variables_Table[CTS_Subsystem_EPSS]
```

DeflateAwning, 2024-06-11:
Triple pointers are rarely the way haha 

Let's just put the subsystem part in with the string part. No need to have extra fields, imo (although feel free to share if I'm missing something :) ).

Using enums as numerical indexes to an array is a little quirky. Not the end of the world, but a little quirky nonetheless. The main issue with it is that there's a risk with out-of-bounds array access if someone adds an element but doesn't update the array, reorders the enum, etc. Seems that Stackoverflow answers disagree on whether it's good or mediocre.

Here's my proposal:

```c
typedef struct {
    char *variable_name;
    void *variable_pointer; // pointer to a uint8_t, uint16_t, etc.
    uint8_t width_bytes; // width of the config variable (e.g., 1 for uint8_t, 2 for uint16_t, etc.)
} CONFIG_integer_config_entry_t;

typedef struct {
    char *variable_name;
    char *variable_pointer;
    uint8_t max_length; // max length, including the null terminator
} CONFIG_string_config_entry_t;

// TODO: something about enum configurations, which are actually the main one we want to use where possible; needs "enum_to_str" and "str_to_enum" conversion mechanisms, probably directly in the command handler?
```

NuclearTea, 2024-06-14:
Updated split string by delim function to modify input string and return first instance before delim

NuclearTea, 2024-06-15:
Created a Telecommand that prints table of all integer and string configuration variables:
[image]

Need to implement set configuration variable TCMD,

and get a single configuration variable TCMD

DeflateAwning, 2024-06-15:
Solid work! 

### Review
1. Bad prototype `uint16_t CONFIG_print_integer_table(char *result)`. You need to specify the `result_len` as an arg to the function, otherwise it doesn't know how long the result buffer is.
2. This doesn't do what you think it does: `strcat(result, "\0");`
3. Configuration.h include guards are non-compliant.
4. You don't need to include `CTS1` in variable names; is self-evident.
5. Rebase.
6. I think the `TCMD_extract_string_arg` function might be better than using the `split_string_by_delimiter` function, depending on what you're trying to do.
7. `split_string_by_delimiter` is missing the `GEN_` namespace prefix.
8. I think returning this as ndjson (aka jsonl) might be pretty good? For example:

```json
{"type": "uint8_t", "num": 0, "name": "test_integer_config_eps", value: 1}
// same for string
```

NuclearTea, 2024-07-12:
Back from Vacation, had to catch up and help unblock other members as well as do interviews

Updates:

- Had to create new branch which was updated with main, there were too many conflicts with my previous branch and so it became really difficult to deal with rebase conflicts
- I also had formatting issues as I have a C/C++ formatter installed which would have messed up the formatting for everyone, I have resolved that in the current branch
- I will continue to work on the above points

NuclearTea, 2024-07-15:
- I will be removing the split string by delim function as it is completely unnecessary
- I am able to set and get int and str config variables

**Note**

Need to revisit how number config variables are handled because we are unable to dynamically allocate memory. In order to have 1 data structure that supports all types of numbers, I currently use a void * however, this is reliant on the default size of void (which in our case, I believe is 4 bytes) and such, numbers above uint32 don't work too well. The easy solution is to make all number config variables the same size (maybe uint32?)

Can remove print table functions as they will be replaced by jsonl function

Need to do:
print data using jsonl function provided by Parker (should be straightforward)

closed 2024-07-27 by NuclearTea