CalgaryToSpace/CTS-SAT-1-OBC-Firmware #18: MPI over-the-air firmware upgrade

DeflateAwning, 2024-03-10:
### Telecommand 1: uplink chunk of new MPI firmware
* Arg 1: start address into file of the MPI firmware
* Arg 2: 256 bytes of MPI firmware
* Arg 3: filename of MPI firmware
* Arg 4: total filesize of MPI firmware

Function must write the incoming data to the "arg 2" byte address in a file in the filesystem.

### Telecommand 2: check hash of new MPI firmware
* Arg 1: intended hash of new MPI firmware
* Arg 2: filename of new MPI firmware

### Telecommand 3: flash MPI with a specific MPI firmware file
* Arg 1: intended hash of new MPI firmware
* Arg 2: filename of new MPI firmware to flash

This telecommand must also verify the hash before flashing.

JohnathanBurchill, 2024-06-04:
Wondering how to write MPI firmware bytes to a specific location in a file in the filesystem. 

Looking at the ```ifs``` functions, the write and seek functions zero-pad if the requested position is beyond the current file size.

Would this be sensible?

For every page,
1. ```lfs_file_open``` for appending
1. ```lfs_file_seek``` to the target address
1. ```lfs_file_write``` the transmitted bytes
1. ```lfs_file_close```, and, possibly ```lfs_file_sync```

Prior to flashing to the MPI, ```lfs_file_truncate``` to the target file size (rather than doing this for each page). Or do this first, in which case the first telecommand would be for target address 0.

DeflateAwning, 2024-06-16:
Totally out of curiousity, did your work-in-progress solution read the response from the MPI after each page/chunk was transmitted, or did it just "fire-hose" data towards the MPI?

JohnathanBurchill, 2024-06-16:
I'm reading the text response each time. Eventually we should just check for a hash response. This so far is between the terminal and the dev kit, no MPI.

I've gotten writing to memory working, sort of. The first page gets written, then the dev kit hangs on the second one. LFS writing is very slow. Needs troubleshooting.

DeflateAwning, 2024-06-17:
Good insight! I'll investigate the LFS speed issue (I've noticed it too); just created Issue #57 to track the progress of the LFS performance assessment and improvements.

DeflateAwning, 2025-06-01:
Just updated this issue to remove/hide the parts of this puzzle that already exist. The scope is now just limited to reading the file out of LittleFS and sending it over to the MPI. 

I'm going to implement this right now. It will just firehose the data to the MPI. We can modify it to check responses if necessary later on.

parker-research, 2026-03-19:
In accordance with discussion with Johnathan today, we'll forego implementing this on CTS-SAT-1. The risks are too high now that the MPI is locked inside the satellite.

If absolutely necessary, this upgrade feature could easily be implemented with the `exec_blob` feature during the mission, with more time for testing.