CalgaryToSpace/CTS-SAT-1-OBC-Firmware #19: MPI/GPS/Camera telecommands: send arbitrary bytes to UART peripheral, and read response

DeflateAwning, 2024-03-10:
In case there's functionality of a UART peripheral we forgot to implement, we want to be able to send artibrary configuration commands, etc. to the peripheral (the MPI/GPS/Camera).

DeflateAwning, 2024-06-16:
This is the general-case version of #32 (send raw MPI command). The solution to that command will be extended to this command.

christopherkinyua, 2024-10-04:
@DeflateAwning I was thinking about the future steps regarding using telecommands to send log commands to the GPS. There was a discussion of having a generic telecommand that would be used to send the log commands to the GPS. I was looking through the issues and this issue could potentially be used in #138. Let me know what you think.

DeflateAwning, 2024-10-04:
This is that generic telecommand, I'd say! This one works in bytes though; it may be helpful to have a plain text version as well.

PR #272 (2025-03-15): Send arbitrary bytes to UART peripheral, and read its response
This is a rebase of https://github.com/CalgaryToSpace/CTS-SAT-1-OBC-Firmware/pull/185, with extra consideration taken for the GPS handler (which supports disabling the interrupt).

Fixes #19.
closed 2025-03-15 by DeflateAwning