CalgaryToSpace/CTS-SAT-1-OBC-Firmware #6: Port LittleFS for this repo

DeflateAwning, 2024-03-10:
We will use [LittleFS](https://github.com/littlefs-project/littlefs) as the flash filesystem for mission data. 

Follow the setup guide. You must implement 4 functions:

- [ ] `user_provided_block_device_read`
- [ ] `user_provided_block_device_prog`
- [ ] `user_provided_block_device_erase`
- [ ] `user_provided_block_device_sync`

After you implement LittleFS and it builds, demonstrate that it works by:
1. Store the boot count in a file named `/boot_count.txt`
2. Write a telecommand to retreive that boot count from the file.

robotoshi, 2024-03-16:
Starting point: 
https://uimeter.com/2018-04-12-Try-LittleFS-on-STM32-and-SPI-Flash/

Saksham-P, 2024-05-29:
## SPI configuration used for Memory Module integration 
[image]

### Changes: 
~~hspi1.Init.DataSize = SPI_DATASIZE_4BIT~~ -> ```hspi1.Init.DataSize = SPI_DATASIZE_8BIT```
~~hspi1.Init.BaudRatePrescaler = SPI_BAUDRATEPRESCALER_2~~ -> ```hspi1.Init.BaudRatePrescaler = SPI_BAUDRATEPRESCALER_8```

Changed Data size from 4Bit to 8Bit and Baud rate pre-scaler from 2 to 8

Saksham-P, 2024-06-01:
### Updates
I have created a branch and put barebones LittleFS code for this revision, with boot_count code in `main.c`

It allows user to do the following (Through coding, not yet through telecommands):
- Create / Open a file
- Write to a already created file
- Read contents of a already created file

### Blockers
None of the functionality is useable through the telecommands yet.
I haven't been able to properly use the UART debug statements on my board (using huart1, which is the only connection I have). This seems like a Rev1.3 IOC issue since in Rev1.1 IOC project, it was working fine.

### Next Steps
Further Improvements to be made:
- Possibly create a global variable for Status Register Buffer in `memory_utilities.c` instead of initializing a local buffer in each function
- Work towards getting the telecommands working (will be a bit slower without having debug available, **but still possible**)
- Develop other core memory functions (line 502 - 733 from `lfs.h` contain other function that can be implemented):
    1. List Directory 
    2. Remove / Delete File
    3. Make Directory

DeflateAwning, 2024-06-01:
Good work! Some feedback:

1. I think the other functions you mention ("List Directory", "Delete File", etc.) are already implemented in LittleFS; they just need telecommand wrappers built for them.
2. I'd advise against a global variable for that purpose, but don't know enough to say for sure.
3. Please ensure all functions you've created conform to the namespace/naming conventions of the repository. Driver functions which access the flash should be prefixed with `FLASH_`, LittleFS functions should be prefixed with `LFS_`, etc. The rest of the function name should be lowercase/snake case.
4. Avoid modifying `main.c`. Put debugging/testing code anywhere else; merging/rebasing main.c with auto-generated code and user code is a bad time :)
5. Please split up `memory_utilities.c`: the flash driver functions, which are meant to be called only from _within_ LFS should go to a separate file called `/littlefs/drivers/flash_driver.c` and `/littlefs/drivers/flash_driver.h` (or similar); the functions meant to be called from just about anywhere in the codebase (e.g., the functions to mount, format, etc.) should go into `/littlefs/littlefs_helpers.c` / `.h`.
6. Rename the chip select functions to indicate that `activate_cs` and `deactivate_cs` are the actions. I'd suggest making the `activate_cs` function take an integer argument, which is the index of the CS pin (there are 8 IIRC, so 0 to 7). Hard code it to 0 for now, but this is the core of how to spread the filesystem across memory banks.

JohnathanBurchill, 2024-06-04:
Does this draft work on the development boards?

JohnathanBurchill, 2024-06-04:
What is the model for accessing a filesystem from a task, like when the telecommand task needs to write MPI firmware bytes to a file?

There seems to be one ```lfs_t lfs``` in ```Core/Src/littlefs/littlefs_helper.c```, and it is used by all the ```lfs_*``` functions. Is this shared amongst tasks?

Saksham-P, 2024-06-04:
> Does this draft work on the development boards?

This should work. The way to use it as of now is to type the configurations of flash memory you are using in `struct lfs_config cfg` in `Core/Src/littlefs/littlefs_helper.c`.
Following this, you should be able to (for now only) write code in `main.c` to test whether you are able to read and write bytes to a file.

> What is the model for accessing a filesystem from a task, like when the telecommand task needs to write MPI firmware bytes to a file?

Currently there are no telecommands or a specific model to access the filesystem from a task. I am currently working on creating flash_memory specific telecommands. 
If a telecommand task needs to write MPI firmware, maybe there will be another call to internal LittleFS functions in `Core/Src/littlefs/littlefs_helper.c`, or maybe call the telecommand for those functions once I implement those if it sounds reasonable.

JohnathanBurchill, 2024-06-05:
Clarified in discussions last evening - STM32 is not running in a pre-emptive mode, so there are no issues with multiple tasks accessing the same variables, files, etc., as long as each task's operation is quick and complete.

PR #45 (2024-06-15): Integrate LittleFS implementation and basic drivers/demo
Closes #6
closed 2024-06-15 by DeflateAwning