CalgaryToSpace/CTS-SAT-1-OBC-Firmware #7: Telecommand: list files in flash

DeflateAwning, 2024-03-10:
Like the Linux `du` command, this command should list the files in flash, ordered by filename (timestamp), descending. 

Arguments:
1. Offset: skip this many files at the start
2. Count: get this many files
3. String contains (optional): a substing that the file path should contain (e.g., filter to only photos)

robotoshi, 2024-07-03:
@DeflateAwning The GNU `du` command lists the size of a given file on the disk.  Is that what you intended?
The description sounds more like a version of `ls` or `find`, so I'm a bit confused.
Is this intended to print out the equivalent of a File Allocation Table?

DeflateAwning, 2024-07-03:
Probably fine to leave out the file sizes for now, unless it's easy to include them

Not too sure what a FAT is, so probably not that. Can you give an example though?

We can tweak it or add more commands if there're operational deficiencies. Just kinda imagine CLI filesystem browsing UX

Saksham-P, 2024-07-03:
The way I am currently thinking of it is basically the `ls` command, but have the files be sorted by filename, and allow for optional arguments such as the ones Parker mentioned:

1. Offset: skip this many files at the start
2. Count: get this many files
3. String contains (optional): a substring that the file path should contain (e.g., filter to only photos, only logs, etc.)

The size of the file shouldn't be too big of an issue to deal with, but I will take baby steps

DeflateAwning, 2024-07-06:
We don't have a mechanism for optional arguments. The "optional" argument must be explicitly specified (e.g., Offset arg will just be 0). Perhaps an empty string can be used to set the string filter argument to "no contains filter".

Saksham-P, 2024-10-16:
Just as an update

The current progress on this is that I have created `list_directory` and `make_directory` telecommands that successfully work, although they only take in 1 argument for now which is the directory name. 
- One thing to note is that currently the `list_directory` telecommand is currently just using `DEBUG` statements to print it to the serial, and the function automatically lists the files/directories in alphabetical order (ascending).

I will work towards getting the arguments `offset` and `count` working next.

Makiv1, 2024-11-17:
I have tested this branch and the code seems to be working correctly.
First I ran `fs_list_directory(,0,10)` which returned the following:
```
========================= UART Telecommand Received =========================
CTS1+fs_list_directory(,0,10)!

=========================
Parsed telecommand (len=32): 'CTS1+fs_list_directory(,0,10)!
'
0000000000000+0000233437_N [TELECOMMAND:NORMAL]: Executing telecommand from agenda slot 0, sent at tssent=0, scheduled for tsexec=0.
========================= Executing telecommand 'fs_list_directory'=========================
Name 	 bytes
.
..
lifecycle

Successfully Listed Directory Contents.
0000000000000+0000233608_N [LFS:NORMAL]: Successfully listed contents from directory: 
========================= Response (duration=158ms, err=0) =========================

===========================================================================

```
This lines up with my expectation as I have created a directory called lifecycle while working on another issue.

Then running `fs_make_directory(test_dir)` **twice** gives the following:
```
Parsed telecommand (len=35): 'CTS1+fs_make_directory(test_dir)!
'
0000000000000+0000524074_N [TELECOMMAND:NORMAL]: Executing telecommand from agenda slot 0, sent at tssent=0, scheduled for tsexec=0.
========================= Executing telecommand 'fs_make_directory'=========================
0000000000000+0000524580_N [LFS:NORMAL]: Successfully created directory: test_dir
========================= Response (duration=494ms, err=0) =========================

===========================================================================
FrontierSat time: 19700101T00:08:44.603, Uptime: 524603 ms
FrontierSat time: 19700101T00:08:45.598, Uptime: 525598 ms
FrontierSat time: 19700101T00:08:46.593, Uptime: 526593 ms
FrontierSat time: 19700101T00:08:47.588, Uptime: 527588 ms
========================= UART Telecommand Received =========================
CTS1+fs_make_directory(test_dir)!

=========================
Parsed telecommand (len=35): 'CTS1+fs_make_directory(test_dir)!
'
0000000000000+0000528223_N [TELECOMMAND:NORMAL]: Executing telecommand from agenda slot 0, sent at tssent=0, scheduled for tsexec=0.
========================= Executing telecommand 'fs_make_directory'=========================
0000000000000+0000528353_N [LFS:WARNING]: Directory test_dir already exists.
========================= Response (duration=117ms, err=1 !!!!!! ERROR !!!!!!) =========================
LittleFS Make Directory Error: -17

===========================================================================
```  

The first time, directory creation succeeds and is confirmed by calling `fs_list_directory(,0,10)`. The second time it fails as the directory already exists. This seems correct.

Finally by calling `fs_list_directory(lifecycle,0,10)` we list the folders inside of this directory and get:
```
========================= UART Telecommand Received =========================
CTS1+fs_list_directory(lifecycle,0,10)!

=========================
Parsed telecommand (len=41): 'CTS1+fs_list_directory(lifecycle,0,10)!
'
0000000000000+0000875407_N [TELECOMMAND:NORMAL]: Executing telecommand from agenda slot 0, sent at tssent=0, scheduled for tsexec=0.
========================= Executing telecommand 'fs_list_directory'=========================
Name 	 bytes
.
..
deploy_antenna_on_boot_enabled.bool	1 bytes

Successfully Listed Directory Contents.
0000000000000+0000875775_N [LFS:NORMAL]: Successfully listed contents from directory: lifecycle
========================= Response (duration=357ms, err=0) =========================

===========================================================================
```

This also seems correct as I have created deploy_antenna_on_boot_enabled.bool while working on another issue.

I have also tried listing a directory which didn't exist: This Failed with error -2.

### Some Closing Thoughts
* code seems to be working correctly when tested on some common scenarios
* This is super useful for when working with the file system!
*  perhaps the response could be a bit more descriptive (logs could label which entries are files and which are directories. errors could be a bit more verbose, avoiding the need for error code lookups in the lfs.h file). Although the code in its current state resolves the linked issue

Great work this is super sweet! Hope this helps!

closed 2025-02-08 by Saksham-P