CalgaryToSpace/CTS-SAT-1-OBC-Firmware #8: Telecommand: take a photo

DeflateAwning, 2024-03-10:
Command should take a photo and store it to the FS.

Args:
1. Light setting

* Please review the camera datasheet spec to confirm if there are any other settings which can be args.

MrBear101, 2024-06-04:
Additional Arg:

    2. Flash setting
 
Current Progress:

    Camera was successfully interfaced with STM32 board.
    Command was sent and Image was received in it's entirety several times through UART.
    Currently merging camera code with a clone of the Firmware repo.
    
Next Steps:

    Will begin interfacing camera with memory module and testing storage of photos using littleFS

DeflateAwning, 2024-06-05:
Can you please get your code in a branch in this repo, and ensure you've pushed that branch?

DeflateAwning, 2024-06-16:
@MrBear101, you should be able to start working on this pretty easily! The filesystem code is in the `main` branch now, so it's just a matter of calling into those functions! Let me or Sakshum know if you have any issues with the filesystem! 

If you need any IOC configuration changes (e.g., setting interrupts or DMA), let me know! Make sure not to commit those changes to your branch.

Hoping to see some progress on this project this week!

DeflateAwning, 2024-08-17:
@MrBear101, can you please provide an update on the progress of this task?

Ozone-Whiskey, 2025-02-28:
@MrBear101 How is thing going?

robotoshi, 2025-04-09:
Ali and Asjad tested and I talked with them today.

@MrBear101 @NuclearTea

We need to add a new RTOS task that handles camera DMA data (and possibly other data in large volumes).
Priority should be higher than "normal".
Should run a `while(1)` loop that polls a volatile flag which signals which half of the DMA buffer is full, if any, and write that data to LFS.

The end of the loop will have an `osDelay()`, the duration of which needs to be no greater than `(time_to_fill_both_halves_of_the_DMA_buffer - time_to_write_a_half_buffer_to_LFS)`.

DeflateAwning, 2025-05-12:
Implemented in #384. Great!

closed 2025-05-12 by DeflateAwning