CalgaryToSpace/CTS-SAT-1-OBC-Firmware #9: Create logging-to-filesystem system

DeflateAwning, 2024-03-10:
* Create a function which can log strings to log files
* Rotate log files, deleting old ones, and avoiding files getting to large

Logs should include:
1. Timestamp
2. System uptime
3. Message

For now, just consider each log a string. Future consideration: we may want to store certain common logs as structs to optimize storage space.

DeflateAwning, 2024-06-11:
We should take a look at uLog: https://github.com/rdpoor/ulog

It's very light-weight, minimal dependencies, and it's intended for embedded systems. Supports creating our own custom log adapter, and could be used to change the verbosities of various different logs. 

The biggest benefit above rolling our own logger imo is that it supports `printf`-style arguments, so we don't need to write to a temporary CString array before calling the logger

JohnathanBurchill, 2024-06-25:
> We should take a look at uLog: https://github.com/rdpoor/ulog
>
> It's very light-weight, minimal dependencies, and it's intended for embedded systems. Supports creating our own custom log adapter, and [...]

Drafted something based on uLog [on this branch](https://github.com/CalgaryToSpace/CTS-SAT-1-OBC-Firmware/tree/burchill-logging-i9). Not issuing a pull request pending a decision about uLog, and implementing log rotation.

uLog unfortunately breaks some of the NASA coding guidelines, such as using elaborate macros, and using function pointers. It does look promising otherwise, and we can adapt it to FrontierSat needs. Both macros and function pointers could be replaced if warranted.

Maybe have dedicated log files like `syslog`, say in `/logs/obc.log`, '/logs/gpc.log', `/logs/camera.log`, etc.

DeflateAwning, 2024-06-25:
Hmm I saw a few different logging options. An informal trade study looking at a few options might be waranted. Thanks for looking into the uLog internals!

I think the main benefit of these logging options is mostly just getting the printf-like syntax for forming strings from variables. If they're way more complicated than just making our own system and using temporary buffers, that might just be better.

DeflateAwning, 2024-06-25:
Just edited the initial comment with a description of the places the log messages should go!

DeflateAwning, 2024-07-13:
I reviewed the entirety of the code, and had these thoughts:

* The use of function pointers/macros appear rather standard and easy-to-reason-about.  If we choose to in the future, it would be easy to replace the function pointer part with basic sequential function calls (although modifying library code _may_ be a worse idea).
* The only elaborate macros I see are of the style: `#define ULOG_INFO(...) ulog_message(ULOG_INFO_LEVEL, __VA_ARGS__)`, which I believe is reasonably standard. We could also just remove those macros and use calls to `ulog_message(...)` directly with the log level enum, if we really wanted to avoid macros.
* I really like that the `ulog_message(log_level, printf-like_format_string)` function writes to an intermediate cstring allocated in the data segment statically (instead of allocating on the stack, as we currently generally do for log messages). This increases the simplicity of performing logging, and allows us to remove the bulk of the variable-length arrays (VLAs) in the code currently. 
* Overall, uLog seems to be quite straight-forward code (not many lines, easy to follow). Logging isn't hard, so I'm glad this library gives a nice foundation without too much complexity. 

After reviewing the uLog code, I am in favor of continuing implementation with this library.

JohnathanBurchill, 2024-07-14:
Okay, will prepare a PR.

Question about arguments: in `littlefs_helper.c`, for example:

```
int8_t LFS_read_file(const char file_name[], uint8_t *read_buffer, uint32_t read_buffer_len)
```

I'm wondering about the rationale for using `[]` for the file_name and leaving `read_buffer` as a pointer, instead of like this:

```
int8_t LFS_read_file(const char file_name[], uint8_t read_buffer[])
```

The function could then use ` uint32_t read_buffer_len = sizeof(read_buffer);`

DeflateAwning, 2024-07-14:
Thanks for the PR! Taking a look now.

Regarding the question about the array args/sizeof, neither syntax "communicates"/passes the size of the array into the called function. Here's a demo I wrote in an online C compiler to demonstrate: https://onlinegdb.com/VnRut6wmu. You'll see that, when executed, both ways yield `sizeof(arg)=8 bytes` (suggesting the server is a 64-bit computer).

Interestingly though, `sizeof(cstring_arg_with_array_syntax)` throws a compiler warning, while `sizeof(cstring_arg_with_pointer_syntax)` returns the sizeof silently. Using the "array syntax" is a good way to ensure the compiler warns in that case, and is a positive change! 

The two options are mostly a matter of style, and are equivalent within the compiler (as far as I know). I generally prefer to use the array-style to communicate that it's an array, unless it's a pointer. No consistency has been enforced, and I frequently forget that personal prefence of mine as I write code, hence the random switching. 

I support the change you made in the PR switching to the array-style syntax.

JohnathanBurchill, 2024-07-14:
Thanks for the info!

closed 2024-07-22 by JohnathanBurchill