CalgaryToSpace/CTS-SAT-1-OBC-Firmware #25: Implement Over-the-Air (OTA) firmware upgrade

DeflateAwning, 2024-05-18:


JohnathanBurchill, 2024-06-05:
@DeflateAwning asked me to comment about a suggestion to look at xboot: 

https://github.com/alexforencich/xboot?tab=readme-ov-file

It's for ATMEL chips, but might provide some ideas or could be ported to STM32?

JohnathanBurchill, 2024-06-05:
The MPI uses it, and I've found it to be reliable. An old version of the MPI had two PCBs each with a microcontroller, and we could do OTA updates on both boards, with one as a passthrough to the other. That was a little less reliable though.

JohnathanBurchill, 2024-06-19:
Can receive memory pages and write to a file via telecommand. Will wait for Issue #47 to be pulled to main to complete uploading to the MPI.

NuclearTea, 2025-04-05:
After investigating this for a bit, there are a couple of things that we have noticed.

1. The current firmware is approx. ~280 Kilobytes in size (this is the release build).
The firmware version will be a huge challenge (especially if we are planning to store multiple copies or if we plan to keep one copy but replace the entire file every time we would like to update it.

There were a couple of ideas that arose from this that @robotoshi, @2BlackCats and myself were discussing.

Toshi mentioned that we can use a lossless compression algorithm to uplink less data and from his initial testing, he has reported ~18% compression (I believe this means that the new size is 82% of the original, please confirm this @robotoshi).

Additionally, Alex suggested that maybe we should not uplink the entire file and rather, use some sort of paging, or diffing algorithm on the ground station and only update the portions of the firmware that would change. This seems like a good idea however, implementing this is quite tricky and I don't think that it's something that we should hand implement (maybe there is a library that can do this for us?). Additionally, we would need ensure that we have an existing copy of the firmware in the memory module.

2. Uplinking the data is a much more difficult (for testing) then actually loading, swapping, and running the new firmware.
My current plan is the following:

- Use the cube programmer to upload the new firmware into the second internal flash bank.
- Load the firmware from the second internal flash bank into FLASH memory
- clear the second internal flash bank
- Run tcmd that takes file that we just created and load it into the second flash bank
- Run tcmd to switch to second flash bank

The above steps will test the following: **Given that there is a file in the FLASH memory, load it into the internal flash, and then switch.**

What needs to be figured out: 
- How to uplink the data in a reasonable manner
- Do we need to upload the new firmware all at once, or use some strategy like paging or diffing
- Which compression algorithm to use
- Suppose We are already running the new program, how do we update the firmware again? (do we have to alter what address we write to in the internal flash?, switching flash banks works as expected?, etc...) -> I will take the lead on this

@DeflateAwning @JohnathanBurchill @robotoshi Would love your input on this.

robotoshi, 2025-04-06:
## Compression

I have been looking at a few options for compression algorithms and implementations.

### LZMA

- Used by linux command `xz`
- `xz`'s implementation seems to require minimum of 1MiB of memory
  - Might be a way to reduce that by fiddling with settings
- [FastLZ](https://github.com/ariya/FastLZ)

### DEFLATE

- Basically what `gzip` uses
- Is very well set up for stream decompression
- A few light/embedded implementations 
  - [uzlib](https://github.com/pfalcon/uzlib/)
    - I got 24% reduction
  - [zlib-deflate-nostdlib](https://github.com/derf/zlib-deflate-nostdlib)
    - Seems *very* lightweight, points to uzlib and udeflate for our use case
  - (udeflate)[https://github.com/jlublin/udeflate]
    - clearly says it only uses stack memory
  - More to investigate 
 
### zstd

- Improved version of DEFLATE 
- [zstd](https://github.com/facebook/zstd/wiki/Using-libzstd-in-a-memory-constrained-environment)

### Heatshrink

- Algo specifically for low overhead
- I get 18% reduction when using their "embedded" settings
- 22% when using non-embedded settings (need to check actual memory usage)

## Patch/Diff

Needs more investigation
- [HPatchLite] (https://github.com/sisong/HPatchLite)

This is a work in progress. I will update this comment with more findings and test results

DeflateAwning, 2025-04-07:
Hmm uplinking 280 KB at 9600 baud should take 4 minutes. Even if a max duty cycle restriction requires us to uplink commands at 1/10th that speed, that's only 40 minutes (3 passes/3 days). 

It seems that pre-`libcsp`, the firmware was only 92640 bytes (e.g., at 1f8404d8). We're on track to be able to remove libcsp, decreasing the firmware size by 65%. 

Looking into these seems cool, but it doesn't need to be/can't/shouldn't be top priority atm. We need to have v1 working and meeting all requirements before we get into these.