CalgaryToSpace/CTS-SAT-1-OBC-Firmware #37: Create automatic safe-mode system

DeflateAwning, 2024-06-09:
Goal: if the satellite is low on power, disable non-critical systems.

### Systems

#### Critical System (Leave Enabled)
1. EPS
2. OBC
3. Comms

#### Systems to Disable
1. GPS
2. ADCS
3. MPI
4. Camera
5. Boom deployment

**This task is waiting on the other systems to be implemented at least in part.**

JohnathanBurchill, 2024-06-25:
Is there a plan for a watchdog timer? Could reboot into this safe mode too if the WDT does not get reset on time due to the processor locking up, say on an assert fail, or if a task fails to `osDelay`.

DeflateAwning, 2024-06-25:
Absolutely on both of those. Made issues #74 (watchdog) and #75 (working assertions) to track each of those.

DeflateAwning, 2025-03-19:
GPS does not have switchable power rails from the EPS. GPS must be put into low power state.

NuclearTea, 2025-03-19:
Sounds like I need to wait for Celina to finish her's for the GPS stuff. Otherwise, disabling the EPS channels for the other peripherals is ok? I was thinking something akin to the following:
```C
            EPS_set_channel_enabled(EPS_CHANNEL_12V_BOOM, 0);
            EPS_set_channel_enabled(EPS_CHANNEL_12V_MPI, 0);
            EPS_set_channel_enabled(EPS_CHANNEL_3V3_CAMERA, 0);

            // this is connected on the same channel as ADCS and GPS
            EPS_set_channel_enabled(EPS_CHANNEL_5V_MPI, 0); 

            LFS_unmount();
```

NuclearTea, 2025-03-20:
Waiting until #301 and clarification for other peripherals.

DeflateAwning, 2025-03-21:
Feel free to just make a stub with a FIXME and mention it in that issue.

DeemDeem52, 2025-03-22:
Come to think of it, I'm not quite sure why we need a specific telecommand for #301 (though I have created it if necessary) -- surely this code itself would do for placement in this function?
`
uint8_t adcs_cmd_status = ADCS_set_power_control(ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF, ADCS_POWER_SELECT_OFF);
`
(With modifications, of course, for any systems which we determine should be kept on, or else kept the same)

DeflateAwning, 2025-03-28:
A specific command is good because, in an urgent low-power situation, being able to immediately command systems to turn off is helpful. The person commanding the satellite at 3am may not have an intricate understanding of all commands available, as well as the arguments supported/required, and only has 10 minutes to figure out how to make the satellite have stable power again.