CalgaryToSpace/CTS-SAT-1-OBC-Firmware #38: Integrate a SHA256 hash algo

DeflateAwning, 2024-06-10:
Create a function called "GEN_sha256_hash(input_arr, input_arr_len, output_arr)", where output_arr is 32 bytes long.

Compute the hash, using an existing implementation.

Create unit tests for short inputs, long inputs, inputs with nulls, and at least 20 random inputs generated with python (or similar).

This will be used in the telecommand uplink validation and authentication mechanism, as well as maybe others.

Benchmark the performance, and leave a comment on how long this takes to compute.

robotoshi, 2024-07-06:
Should use this library:
https://www.st.com/en/embedded-software/x-cube-cryptolib.html

Makiv1, 2024-08-27:
Hi, I've integrated the library on the attached branch. Should I only keep the sha256 header files or keep all of the header files for the other cryptographic functions?

DeflateAwning, 2024-08-27:
I think that linked library is meant to be enabled via the IOC file and the CubeMX code generation. Either's fine for now!

DeflateAwning, 2024-10-22:
Finally got the ST_Cryptolib library enabled in the `main` branch (in 603f8e01d6ef35b0b286dce3aaf80f4f5790c34c).

I believe this project is in progress in the `marko-sha256` branch. @Makiv1 can update that branch to use the new library, which will close this issue.

DeflateAwning, 2024-10-25:
After struggling through with STMicroelectronics libraries which contain pre-compiled mystery chunks, and the SAFEA1 library, which contains paths with `#` in them (which destroy Makefiles), I recommend using a very simple C implementation, like this one: https://github.com/B-Con/crypto-algorithms.

For an example of how to use that, see: https://github.com/B-Con/crypto-algorithms/pull/9/files

I recommend applying the change in this PR here: https://github.com/B-Con/crypto-algorithms/pull/21/files

Thanks!

robotoshi, 2024-10-25:
Another library recommendation.  It's optimized for Cortex-M

https://github.com/ctz/cifra/

PR #223 (2024-10-30): Implement SHA256 hash function
resolves #38 

* Implements sha256 hash function which hashes a full message
* includes support for hashing chunks at a time instead of all in one go
* 20 random 4 killobyte alpha-numeric messages have been generated along with their expected hash and used to create 20 unit tests
* These unit tests are benchmarked when ran and the results logged
* Hashing 20 4 kilobyte messages takes ~490ms

sha256 implementation was taken from: https://github.com/B-Con/crypto-algorithms
closed 2024-10-30 by Makiv1