CalgaryToSpace/CTS-SAT-1-OBC-Firmware #39: Integrate a CRC32 checksum function

DeflateAwning, 2024-06-10:
Create a function called "GEN_crc32_checksum(input_arr, input_arr_len, output_arr)". Specify the length of the output array in the comments.

Compute the checksum, using an existing implementation. Consider using the special hardware on the STM32, if you can.

Create unit tests for short inputs, long inputs, inputs with nulls, and at least 20 random inputs generated with python (or similar).

This may be used in the telecommand uplink validation mechanism, in validating the data coming from the MPI, and maybe other locations too.

DeflateAwning, 2024-07-06:
Here are the options to configure in the STM32 CubeMX program:
[image]. I've prepared to set it to the default values in the upcoming PR. Let me know if we need to set any values specifically.

DeflateAwning, 2024-07-13:
The `main` branch now contains the files to use the CRC HAL functions.

DeflateAwning, 2024-08-16:
@JohnathanBurchill, would you mind adding a couple test cases of the CRC values the MPI uses? Just want to make sure we're getting the polynomial correct for it

JohnathanBurchill, 2024-08-22:
> @JohnathanBurchill, would you mind adding a couple test cases of the CRC values the MPI uses? Just want to make sure we're getting the polynomial correct for it

Sure, please get back to me in mid Sept - super busy and this seems low priority, but let me know if not :)

NuclearTea, 2024-10-22:
@Ozone-Whiskey Hey Shaan, I was wondering what your progress is for this

closed 2025-02-12 by Ozone-Whiskey