CalgaryToSpace/CTS-SAT-1-OBC-Firmware #54: Create AX100 (comms system) driver functions

DeflateAwning, 2024-06-16:


JohnathanBurchill, 2024-06-25:
Is this the "radio"? As a HAM, I'd be interested to assist. :)

DeflateAwning, 2024-06-25:
Is indeed the radio! Very awesome; this is definitely the largest and most significant piece of the firmware that hasn't received adequate attention. I'm sure the Comms team would love to help getting going with it

JohnathanBurchill, 2024-06-25:
@pshme Would be great to chat with you about programming the radio.

Makiv1, 2025-01-29:
Going to pick this up now as the antenna deployment drivers are done

Makiv1, 2025-03-28:
At this point there are drivers for downlinking up to 220bytes of data. In order to transmit more than that we'll have to fragment packets and reassemble them on the ground (could use/implement RDP or something similar). Thoughts on next steps?

DeflateAwning, 2025-03-29:
Have we figured out where that 220 byte limitation comes from? The [CSP wikipedia page](https://en.wikipedia.org/wiki/Cubesat_Space_Protocol) suggestions that we should be able to send way longer packets, so just wanted to pinpoint where that's coming from.

Makiv1, 2025-03-29:
Limit is from the ax 100, chose 220 arbitrarily since limit will depend on which settings we use with the ax100. For reference [image]

DeflateAwning, 2025-04-26:
Closing as these features are getting implemented in #55, #98.

closed 2025-04-26 by DeflateAwning