CalgaryToSpace/CTS-SAT-1-OBC-Firmware #67: LFS: Implement a way to create inline files

Saksham-P, 2024-06-21:
Files need to be created inline to save memory in the memory module. Creating the files without them being inline would cause one file to take an entire block of space in memory. 

For e.g. 
* if `BLOCK_SIZE = 262144` in bytes, then a file that contains `512` bytes of data would consume and hold `262144` bytes of space. 
* To create a inline file in LittleFS, we might need to use `LFS_TYPE_INLINESTRUCT = 0x201` flag when creating a file.

DeflateAwning, 2025-05-06:
Is this issue still required? Is it beneficial to implement? 

If we do want it, we need to do it very fast so that we can ensure timing requirements are maintained (in case this functionality slows things down at all)

Saksham-P, 2025-05-07:
This would require a bit of testing after it's initial implementation. I think the need for this functionality comes down to whether we will need more than 2048 files throughout the mission. Which in my opinion we may need more.

I could get started on this, and I do wanna ensure that we would only consider using inline for small files that won't need to be changed later. Because since inline files are part of metadata, updating them would cause a time overhead since littlefs may need to restructure the entire metadata. 

There is also a max file size for inline files which has to be <= `block_size/8` or <= `cache_size`, whichever one is smaller. In our case, it would be the `cache_size` since we have it set to 2048 bytes. We could increase that to match `block_size/8`, which would increase the max inline file size to be 16384 bytes, which should be enough for quite a few type of files.

Let me know your thoughts on this, whether you think we should go ahead with this.

Saksham-P, 2025-05-08:
For each file that is created in littlefs:
- If `file_size` <= `cache_size` the file will be created in metadata (inline)
- if `file_size` > `cache_size` a whole block will be allocated for the file (regular)
- If `file_size` changes to be > `chache_size` it will be moved outside of the metadata into a block (regular)

To utilize inline efficiently, we need to find the correct balance of `file_size` such that we don't create too much time overhead with many inline files