CalgaryToSpace/CTS-SAT-1-OBC-Firmware #68: LFS: Create enum for all possible errors that can occur in `flash_driver.c` to group them in certain categories

Saksham-P, 2024-06-21:
Enum could go in `flash_driver.h` where all possible return errors such as `FLASH_SPI_TRANSMIT_FAILED`, `FLASH_SPI_RECIEVE_FAILED`, `FLASH_TIMEOUT`, `FLASH_OK` ... & more.

This would allow error handling to be more straight forward and meaningful when it comes to telecommands.

DeflateAwning, 2024-06-21:
This is definitely a very standard/common way of propogating errors (instead of returning random non-zero numbers as we currently do). 

However, I think the value-add for this project is rather low; the main reasoning is that the functions that call the flash drivers don't really care what failed, and don't do any conditional logic based on the failure type. That's the main benefit to using this style of return is to enable strict conditional logic for different failure modes.

What we currently do primarily (dare I say, instead) is have logging within each failure check (right before the non-zero return). This is far more valuable for debugging imo. Enabling remotely-accessible logging is another issue (#9, #30), which we'll absolutely do. 

An alternative to error enums would be to define the error states in a general error definitions file. This is more like how LittleFS does it. Not sure what the best plan is.

```c
#define ERROR_FLASH_WRITE_ENABLE_LATCH_TIMEOUT 1
#define ERROR_FLASH_READ_TIMEOUT 2
```

Anyway, low priority overall imo. We already have 95% of the value: non-zero returns, which are checked after every call and propogate upwards.

JohnathanBurchill, 2024-06-24:
Based on the LFS_ASSERT trouble in #70, error values returned in the four flash functions (read, write, etc.) whose function pointers are stored in `lfs_cfg` must be less than or equal to 0.

It's a matter of style, though I do like the idea of an enum declared in flash_driver.h to cover any errors returned by those functions. Like,

```
enum FLASH_ERRORS {
    FLASH_ERR_OK = 0,
    FLASH_SPI_TRANSMIT_FAILED = -1,
    ...
};
```

DeflateAwning, 2024-06-24:
I think that makes sense to do! No objections; should be straightforward to group/categorized the errors if someone wants to

Saksham-P, 2024-06-26:
I could get started with making sure that the returned error values are < 0 since that seems like an issue that could affect LittleFS greatly.

While doing this though, I think I can also just work on the categorization as it wouldn't take much longer to add this on

DeflateAwning, 2024-06-26:
Go for it! Thanks

closed 2024-07-13 by Saksham-P