CalgaryToSpace/CTS-SAT-1-OBC-Firmware #70: LFS_ASSERT failure

JohnathanBurchill, 2024-06-23:
LFS_mount() fails at

```
Line 119        LFS_ASSERT(err <= 0);
```

in `littlefs/lfs.c`. This appears to arise from the FLASH_... read function returning error values greater than zero. The above assertion failure was in response to this error in `flash_driver.c`:

```
Line 381    const HAL_StatusTypeDef rx_result_1 = HAL_SPI_Receive(hspi, (uint8_t *)rx_buffer, rx_buffer_len, FLASH_HAL_TIMEOUT_MS);
    if (rx_result_1 != HAL_OK) {
        FLASH_deactivate_chip_select();
        return 3;
    }
```

This is from the main branch as updated yesterday with commit `407a78fab48120e8c2c39`.

JohnathanBurchill, 2024-06-23:
LFS_mount() works when commit `407a78fab48120e8c2c39` is reverted; maybe not a hardware issue.

DeflateAwning, 2024-06-23:
My hunch is that one of the timeout variables (probably `FLASH_HAL_TIMEOUT_MS`, if the error is the `HAL` return value) is now set too low, based on the characteristics above. If you're interested in debugging it, would you mind trying increasing `FLASH_HAL_TIMEOUT_MS` from 5ms to, say, 10ms, or 100ms for debugging? 

Reading 512 bytes at 2 Mb/s (the current SPI speed configured) should take 2 ms, so the current 5 ms timeout _should_ be enough in theory. 

@Saksham-P, are you having this problem when running what's in `main`?

JohnathanBurchill, 2024-06-23:
Will check timing.

The main issue is that Flash and LFS designs are currently inconsistent in terms of passing return values from Flash driver to LFS.

JohnathanBurchill, 2024-06-23:
Setting `FLASH_HAL_TIMEOUT_MS` to 6 was sufficient, but leaving it at 10 may be wise. There may be a lot of overhead in the function calls, etc., so the 2 ms estimate may be in the noise for transferring only 512 bytes.

DeflateAwning, 2024-06-23:
See my thoughts (opinion) on return error value "reform" in #68 (tldr: low-value change at the moment). Definitely open to discussion/PRs if anyone sees it differently of course though.

In this case, LFS returns -1 on any non-zero return from the flash drivers. I'm not convinced consistency would offer anything more in this case.

I do think differentiating between timout vs "other" HAL error in that flash driver is a good fix.

JohnathanBurchill, 2024-06-24:
I might be missing something, because the LFS_ASSERT was triggered because the return value from the read function pointer was "3", and the code in lfs.c wanted to assert that it was negative, putting the micro into a "do nothing forever" loop:

In `lfs_bd_read(...)`, starting at line 116:

```
        int err = lfs->cfg->read(lfs->cfg, rcache->block,
                rcache->off, rcache->buffer, rcache->size);
        LFS_ASSERT(err <= 0);
        if (err) {
            return err;
        }
```
Here, `lfs->com->read` is a function pointer to the "user-supplied" `LFS_block_device_read`.

DeflateAwning, 2024-06-24:
You're spot-on there. We should indeed fix those to return negative values. I misunderstood what your suggestion was (as I had that other issue on my mind) :)

PR #71 (2024-06-24): Increase FLASH_HAL_TIMEOUT_MS to 10 ms
Fixes #70 failed read from flash chip.

Does not fix ASSERT issue where FLASH-related function return values are positive, but ifs-related functions assert errors to be less than 0. Suggest to open a new issue to address this specifically.
closed 2024-06-24 by JohnathanBurchill