CalgaryToSpace/CTS-SAT-1-OBC-Firmware #74: Enable software watchdog

DeflateAwning, 2024-06-25:


NuclearTea, 2025-02-12:
@DeflateAwning Activating is straightfoward using the cubeMX. Do you have any specific parameters you would like to modify? The default is the following:

[image]

robotoshi, 2025-02-19:
### IWDG

Independent Watchdog

- Classic watchdog, resets when not pet often enough
- Runs on its own dedicated clock

### WWDG

- Resets if pet too soon, **or** if pet too late
- Runs on the system clock
- Mainly used to detect if the watchdog is being pet in a fast loop and other code is not being run
- Also allows an interrupt to trigger right before it barks, like to cleanup or track what made it hang

### Recommendation

People say to use both, with the IWDG as a last resort backup if the WWDG fails
WWDG should have a much shorter timeout, and IWDG should have a longer timeout.

DeflateAwning, 2025-02-19:
Seems like the range of the Windowed Watchdog is on the order of 1ms to 10ms (max around 20ms). Let's refine the scope of this issue to just configure the IWDG, as I don't really see any uses of the WWDG in our current architecture.

It seems like the WWDG is intended more for very low latency applications like control systems (e.g., negative feedback loop, medical device control, etc.), and less-so for a command-based system like ours where commands/tasks generally block for closer to 50-500ms.

DeflateAwning, 2025-03-15:
Enabling IWDG with timeout of 16.384 seconds:

```
LSI = 32,000 Hz
Prescaler = 32 (IWDG_PRESCALER_32)
Reload = 4095 (max value)

T_timeout = ((RLR + 1) * 4 / LSI_FREQ) * (2 ** PR)
= (4096 * 4 / 32k) * 32 = 16.384 seconds
```

### Regarding the "Window Value" field (in the IWDG)

The window value adds a restriction to when the watchdog can be refreshed.

If Window Mode is enabled, you can only refresh the watchdog when the down counter is BELOW that value.

If you attempt to refresh too early (before reaching the window threshold), the watchdog triggers a reset.

This prevents software from blindly refreshing the watchdog too frequently, which can mask failures.

Suppose RLR = 4095, giving a max timeout of 16.38 seconds.
If window value = 2048, you can only refresh the watchdog after 8.19 seconds have passed (halfway through the countdown).
If you refresh it before that, the watchdog will reset the system immediately.

Thus, setting `RLR = Window_Value`, you can refresh anytime.

For our purposes, we will require at least 0.2 seconds to pass between pets.

```
Window_Value = (16.384 sec - 0.2 sec) * 4096 / (16.384 sec)
= 4046
```

### Testing

1. Try stalling for more than 16 seconds (use `demo_blocking_delay` telecommand).
2. Try calling the watchdog servicer in a loop with no delay. Should cause reset.

During test, pay attention to the reset reason (should be related to watchdog).

DeflateAwning, 2025-03-16:
Done. Pushed in the two commits above ^

commit 6a514a24 (2025-03-15): Add Watchdog servicer to `TASK_execute_telecommands()`

Closes #74
closed 2025-03-15 by DeflateAwning