CalgaryToSpace/CTS-SAT-1-OBC-Firmware #78: Write a driver for the builtin RTC

robotoshi, 2024-06-28:
Should include:
- A function that accepts the current time in the format given by the GPS and sets the RTC to that time and returns the number of milliseconds that the clock was changed by
- Another function that does the same as above but accepts the time as "milleseconds since 1970"
- A function that return the current time of the RTC as "milleseconds since 1970"

NOTE: This will require a change to the .ioc file

Next step is to extend the timekeeping system to use this driver.

JohnathanBurchill, 2024-07-03:
Does the OBC PCB have a backup battery independent of the EPS? Or, say, does the EPS provide an RTC-like backup power line?

I've taken a go at programming the RTC, though the development kit has no backup battery, so cannot test beyond pressing the "reset" button on the board.

DeflateAwning, 2024-07-03:
No backup battery on the OBC; just the EPS supply. 

It just occured to me that the EPS also has an RTC (maybe only 1-second accuracy), so that might be a good option to sync on boot.