CalgaryToSpace/CTS-SAT-1-OBC-Firmware #89: Create function to parse a hex string into a byte array

DeflateAwning, 2024-07-06:
* Needed for Issue #32.
* Needed to finish implementing the `TCMDEXEC_flash_write_hex` function, which currently throws an annoying build warning

```c
uint8_t TCMD_extract_hex_array_arg(const char *args_str, uint8_t arg_index, uint8_t result_array[], uint16_t result_array_size);
```

Function should support ignoring spaces and underscores within the string, which are used as delimeters. 

You may wish to factor out the hex parsing logic to one or more other functions, to separate it from the comma-searching nature of the `TCMD_extract_` functions.

robotoshi, 2024-07-06:
This is a duplicate of #77, but with more details

parker-research, 2024-07-17:
Implemented in #111

closed 2024-07-17 by parker-research