CalgaryToSpace/CTS-SAT-1-OBC-Firmware #91: Upgrade the HAL library

robotoshi, 2024-07-06:
Issue #39 requires the newest HAL library, or at least the full one (for CRC).
Either include the HAL library as a git submodule or use CubeMX to update it.

DeflateAwning, 2024-07-06:
Can you provide a source? We should be on the latest one. Perhaps we just need to enable the CRC hardware support in the IOC file?

Where are we using CRC32 again? Does it need to be fast? Can we just copy-paste in a simple implementation that doesn't require using specialized hardware?

robotoshi, 2024-07-06:
My bad, it is the most recent version.  I didn't check carefully enough before.
CubeMX must not have included the CRC part of it.

Of course it's possible to use another library, but if the functionality is built in to hardware, uses the HAL, and confirmed works on this device, I don't see any reason not to use it.

DeflateAwning, 2024-07-06:
Let's move the CRC discussion back into #39

closed 2024-07-06 by DeflateAwning