CalgaryToSpace/CTS-SAT-1-OBC-Firmware #99: Create temperature sensor driver

parker-research, 2024-07-10:
The OBC has an onboard temperature sensor connected by I2C (via I2C4).

Per the OBC schematic:
```
Model Number: STDS75DS2F
Address: 0b1001_0001 = 0x91
```

Create drivers, and test it in both positive and negative temperatures.

Main deliverable: a function which returns 0 on success, and which puts the temperature into a variable named `result` by a pointer argument.

parker-research, 2024-09-13:
Hey, @parkerg-13, any luck with this? I'm hoping we can have a PR for this merged into `main` in the next 7 days, as it would be helpful for the upcoming testing opportunity.

parkerg-13, 2024-09-13:
Hey @parker-research, I'm pretty close to being able to put a pr up. Just have to clean a couple more things up and test a couple more things. I'll try and have it up for review Saturday so there is time for feedback.

closed 2024-10-30 by parkerg-13