CalgaryToSpace/CTS-SAT-1-OBC-Firmware #123: ADCS: extract and log magnetometer data (if possible)

parker-research, 2024-07-24:
@JohnathanBurchill requested, as potentially beneficial to science mission

Waiting on ADCS Drivers in #13

DeemDeem52, 2024-07-27:
Getting magnetometer values is something I can do at the driver level for the ADCS; indeed, I think that's one of the functions I'm implementing anyway! I'll get back to you with more details.

DeemDeem52, 2024-07-27:
Okay! Looking through the datasheet and code, there are a few commands which will be relevant here:

Implemented in the branch based on commissioning manual:
- Deploy Magnetometer
- Set Mode of Magnetometer Operation
- Set/Get Magnetometer Configuration
- Set/Get Estimation Parameters (magnetometer noise, etc)
- Get Raw Magnetometer Values

Unimplemented (not required by commissioning manual):
- Secondary Magnetometer Raw Measurements
- ADCS Temperatures (temperature of magnetometer)
- Redundant Magnetometer Configuration
- Current ADCS State (whether there's been an error)

@JohnathanBurchill Which of these do you think would be most helpful?

JohnathanBurchill, 2024-07-28:
> 
> @JohnathanBurchill Which of these do you think would be most helpful?

All the implemented functions look useful. The raw mag values would be what I was thinking of. All the unimplemented stuff would be useful too, particularly ADCS state and secondary mag raw measurements.

Does the ADCS documentation specify the measurement resolution / accuracy / precision of the magnetometer? 

What might make sense it to have a `housekeeping_data()` telecommand which would download a log or binary file of the ADCS, GPS, etc. measurements recorded during the MPI operation.

DeflateAwning, 2024-09-04:
@DeemDeem52, which function(s) in your ADCS branch solve this issue? I'm sure they're there; just want to make sure they're explicitly denoted as such.

DeemDeem52, 2024-09-05:
> @DeemDeem52, which function(s) in your ADCS branch solve this issue? I'm sure they're there; just want to make sure they're explicitly denoted as such.

I wrote in my above comment the five functionalities which are explicitly implemented as telecommands; the other ones which I mentioned are not implemented in my pull request -- however, the generic telecommand and telemetry functions should allow use to use those functionalities anyways.

I suppose that means that we can either: 
* close the issue when the PR is merged,
* keep the issue open and merge the PR,
* implement the rest of the functions explicitly (should be pretty easy) and then merge the PR.

DeflateAwning, 2024-09-05:
Thanks, I missed those messages above. I think what I've gathered is that the _really_ useful command/function is probably `ADCS_Get_Raw_Magnetometer_Values`. 

A "log unified GPS/ADCS data useful for MPI data collection" command/periodic action (proposed as `housekeeping_data()` above) will be easy enough to implement; it will simply query the GPS and ADCS several times each with different commands as necessary, keep only the info we want for the mission (to save space), and then log it (probably as JSON). We will save that implementation for another issue though. 

I think this issue is "fixed" by PR #151, because that PR implements the `ADCS_Get_Raw_Magnetometer_Values` function.

closed 2024-09-21 by DeemDeem52