CalgaryToSpace/CTS-SAT-1-OBC-Firmware #134: Ground Support Test: ensure delay telecommand works as expected

parker-research, 2024-07-31:


2BlackCats, 2024-12-01:
Are we expected to create unit test for the "demo_blocking_delay" telecommand? Or am i just trying different args to get a unexpected result?

2BlackCats, 2024-12-02:
@parker-research

parker-research, 2024-12-03:
This will be an external test, where the test is initiated and tracked from a computer connected to the satellite/dev kit via UART. @kev9268 was working on getting the test runner setup in Python, after which we'll add a test for this. 

The idea behind these types of tests is to be able to perform system-level integration tests, validating the telecommand handler and hardware peripherals from outside the satellite itself. If the satellite has a faulty clock which does not delay accurately, for example, then it would also not be able to assess that its clock is ticking significantly fast or slow (in the case of the issue this test describes).