CalgaryToSpace/CTS-SAT-1-OBC-Firmware #146: Antenna Deploy: implement temperature mapping (raw to degrees C)

DeflateAwning, 2024-08-02:
There's some crazy logic for converting the temperature that the antenna reports from the raw value received over I2C (which is a 10-bit ADC value) to a temperature in degrees C.

Reference the user manual to work out how to do this conversion. Consider also finding another library that has the lookup table, instead of copying it out of the PDF.

Test case:
```c
uint8_t raw_bytes[2] = {0x3D, 0x02};
float deg_c = ANT_convert_temperature_raw_to_deg_c(raw_bytes);

// assert between 23 and 24, based on the datasheet
ASSERT_TRUE(deg_c > 23.0);
ASSERT_TRUE(deg_c < 24.0);
```

Ideally, try to get float accuracy, but int accuracy is fine too.

The chip used on the antenna system is `LM94022 with GS=10`, in case you reference a datasheet.

DeflateAwning, 2024-09-04:
@christopherkinyua, any updates on this one?

christopherkinyua, 2024-09-04:
@DeflateAwning  Someone else can pick this up. I was hoping to take this on with my GPS ticket #16 but that's taking a while to finish

DeflateAwning, 2024-09-20:
From datasheet,

```
V_out = 3.3 * Raw / 1023
Temperature_C = f(V_out), where f is the lookup table.
```

#### Test Cases
`Raw = V_out * 1023 / 3.3`

1. 75C -> V_out=1.279V -> Raw = 396.49.
    * Thus, with Raw = 396 -> assert Temp_C is between 75C and 76C.
2. -45C -> V_out=2.571V -> Raw = 797.
    * Thus, with Raw = 797 -> assert Temp_C is between -44.8C and -45.2C
3. 1C -> V_out=2.089V -> Raw = 647.59
    * Thus, with Raw = 647 -> assert Temp_C is between 1C and 2C

DeflateAwning, 2024-09-20:
Datasheet:
[image]

Based on that:

```
const double m = -0.29648;
const double b = 193.375;
Temp_C = m * raw + b;
```

closed 2024-09-25 by Makiv1