CalgaryToSpace/CTS-SAT-1-OBC-Firmware #149: Flash Telecommand: get and print the Status Register for a flash module

DeflateAwning, 2024-08-02:
Super easy command. Would love to get it in `main`. Would be helpful for checking on the general status of flash memory modules during a variety of testing actions.

Would this implementation be the same or different for the NAND vs. NOR modules, @Saksham-P?

Saksham-P, 2024-08-02:
The implementation would be very slightly different for NOR vs. NAND. 

- For the NOR modules, we can simply call a command that would return the values of the Status Register Byte.
- For the NAND modules, we have to call a command that lets the module know we want to access one of it's "features", and as a data byte we send the address of Status Register feature which we want to access. 

Overall, the main differences being apart from the command byte values which differ
- For NOR: 2 Transmit Commands, 1 Receive Command
- FOR NAND: 1 Transmit Command, 1 Receive Command

DeflateAwning, 2024-08-02:
Sweet! Hmm let's focus on just doing this for the NAND module only in its branch then.

Would be awesome to rebase that branch to `main` as well.

PR #212 (2024-10-18): Flash: Added status register telecommand
Resolves #149 

**Summary**
Added a simple telecommand that reads the status register value from the flash memory module and prints as hex value.
closed 2024-10-18 by KaleF07