CalgaryToSpace/CTS-SAT-1-OBC-Firmware #172: TCMDS for Reading and Writing to intern flash memory

NuclearTea, 2024-08-31:
Develop an API to Read and Write to the internal Flash Memory.

Take Specific care to think about the bootloader

DeflateAwning, 2024-09-01:
Is this a telecommand or a function-based API? 

I propose that you should write to a file in the LittleFS filesystem (which I believe we have as a function), verify that data with a checksum, and then copy that to the STM32's flash memory. I think telecommands to read and write to the STM32 flash memory could be helpful for debugging and verification, but I propse labelling them as "not intended for use during mission"

NuclearTea, 2024-09-01:
In order to write the initial golden copy and for testing purposes, it would be more convenient just to have a Telecommand which could write to the internal flash memory directly. It would use functions which would be used by other telecommands. As you stated, this telecommand is not intended to be used during flight operation, more for convenience sake

closed 2024-09-26 by NuclearTea

NuclearTea, 2025-02-12:
@DeflateAwning My new PR #255 Includes this functionality. There is a TCMD to toggle it, and a TCMD to retrieve it. I am still testing out the consequences of switching to dual bank mode as it changes the virtual memory address space and thus, affects writing to the internal flash.

Me and @robotoshi talked about this yesterday and he suggested that the OBC should be in Dual Bank mode by default. I believe this is trivial to do and should be a configuration that can be set in IOC.

DeflateAwning, 2025-02-12:
Toshi seemed to suggest that, with the STM32 in single-bank mode, it's not possible to write to that flash bank while code is running. Does that match your experience? 

My understanding was that, as of this PR (172), the STM32 was in single-bank mode AND you were able to write to flash.