CalgaryToSpace/CTS-SAT-1-OBC-Firmware #175: STM32 ADC: Monitor VBAT voltage with STM32 through voltage divider on OBC

DeflateAwning, 2024-09-04:
* Create telecommand which reports the VBAT voltage

Freddydeddydoo, 2024-11-20:
Where would I put this in the telecommand defs.c? I don't think I should put this in the eps section because I don't think I need whatever external eps module to run this. I could be wrong.

DeflateAwning, 2024-11-21:
EPS telecommands file is probably fine. You'll also want to create a new function called something like `EPS_read_vbat_with_adc`, which could go in its own new file in the eps folder.

Freddydeddydoo, 2024-11-30:
I'm going to put the helper function of this telecommand in the eps_commands.h header file. I don't know if that's the best place to put it because I am just using Hal ADC functions and not really getting anything straight from the EPS. However, this function will only be used for the eps so that's why I'm putting it in there.

DeflateAwning, 2024-12-01:
Seems fine to me! Viable options are either in the EPS folder (maybe in a new file), or in the STM32 folder (e.g., where the "reset reason" files are). Not too concerned about where it goes though!

DeflateAwning, 2025-04-07:
Any luck with this one?

Freddydeddydoo, 2025-04-16:
Yeah, sorry. I did other PRS after this and kind of lost track of it. I will restart this PR because something went wrong, and it wouldn't build, lol. I'll get it done by this week or next.

Freddydeddydoo, 2025-04-23:
Yea I'm starting a new branch. I wrote down all my changes so it should be fine

Freddydeddydoo, 2025-05-02:
I had it. I forgot to clean build -_-

DeflateAwning, 2025-05-10:
Voltage divider equation:

```
vbat_monitor = vbat * (1k / 11k)
```

closed 2025-05-12 by Freddydeddydoo