CalgaryToSpace/CTS-SAT-1-OBC-Firmware #210: ADCS: Repeatedly poll and save ADCS telemetry

DeemDeem52, 2024-10-09:
Now that the main bulk of ADCS command functions are complete, the Commissioning Manual requires that certain ADCS telemetry requests be repeatedly polled every few seconds over the course of between several minutes to one full orbit (see Sections 2.2 and 2.3 of the Commissioning Manual for more details).

In total there are 30 telemetry request functions that must be called in various combinations, of which there are 18 combinations which we care about for commissioning.

This function will need to take inputs:
- Which functions to poll (either the specific functions to poll, or else the commissioning step -- leaning towards the latter)
- The period to poll them over (in seconds; either 1 second or 10 seconds)
- How long to poll them for (either time in minutes/hours or else enable/disable; need to check what's most appropriate for the "1 orbit" condition)

What it will need to do:
- Check the ACP Execution State telemetry to ensure that Time Since Iteration Start > 500 ms (see Section 2.3)
- Call the appropriate telemetry request functions and load them into their structs (the Current ADCS State telemetry request may also be necessary)
- Save the structs to memory (preferably with timestamp, which is another ADCS telemetry request or can be from the OBC)
- Do this all periodically until the time of polling is complete, then downlink the data when next possible.

This downlink will be for ground processing, and varies depending on the commissioning step.

DeemDeem52, 2024-10-09:
Thanks! That's a good point, actually -- this functionality could be pretty useful more generally.

As for the one-second interval... as far as I can tell, all the ADCS data should be logged within one second from the 500 ms Time Since Iteration Start so that it's all on the same frame.

DeflateAwning, 2024-10-09:
What does "frame" mean in this case? Is it some sort of time partition/sampling the ADCS uses? Or more related to data partitioning?

DeemDeem52, 2024-10-09:
The commissioning manual seems to use "frame" to refer to a single update of all the ADCS telemetry... so it would be a time partition.

DeemDeem52, 2024-11-02:
It looks like there's a built-in method to log telemetry data to the ADCS SD card, which generates log files directly. I think it will be easier to do this for the ADCS purpose specifically, because then the ADCS can do it automatically when we request a given commissioning step.

closed 2025-02-05 by DeemDeem52