CalgaryToSpace/CTS-SAT-1-OBC-Firmware #250: Duplicate Telecommands: Should telecommands get added due to @tssent or @tsexc?

Freddydeddydoo, 2024-12-11:


DeflateAwning, 2024-12-11:
`@tsexec` should determine when the command is executed. While it is true that no 2 commands _should_ have the same tsexec timestamp, it's not a big deal if two do, and they will be executed in arbitrary order. 

`@tssent` is intended to be a unique ID to de-duplicate incomming commands. Due to the forward-moving nature of time on earth, no two commands can be sent at exactly the same timestamp, and the `@tssent` timestamp will be added automatically by the ground station transmission program (i.e., no room for human error). Thus, the `@tssent` timestamp uniquely identifies each command, and is perfect for deduplicating. 

Issue #53 correctly suggests using the `@tssent` field, and PR #231 correctly implements it using the `timestamp_sent` field of the incoming command struct. Thus, I see no issues.

I propose this issue can be "closed as not planned", unless you had any other questions or thoughts!

Freddydeddydoo, 2024-12-12:
Okay thanks for the clarification. I was so confused when I reran this telecommand. I will close this issue now.

Freddydeddydoo, 2024-12-12:
Wait sorry I still need to fix my bug which I'll do later today my bad

Freddydeddydoo, 2024-12-13:
@DeflateAwning  sorry could you clarify the question about tssent in the discord? I guess it really isn't a bug sorry I wasn't thinking straight.

DeflateAwning, 2024-12-13:
Please keep discussions related to issues associated with the relevant issue, where possible.

Would you mind reiterating the question here?

Freddydeddydoo, 2024-12-13:
Currently, any command will fail after one telecommand is sent since all telecommands default to 'tssent=0', should I add the conditional in the for loop, for now, to 'fix' the issue as shown below?    
[image]

Freddydeddydoo, 2024-12-13:
I apologize. Let me add additional information. Issue #53 addressed the problem of duplicate telecommands sent simultaneously. I fixed that, then realized that all telecommands currently default to tssent=0 after @KaleF07 addressed an issue about sending telecommands a second time. Currently, I want to solve this by simply adding a conditional to line 53 in [Telecommand_executor.c](https://github.com/CalgaryToSpace/CTS-SAT-1-OBC-Firmware/blob/4498fb1aaa9644ef9e660fcf3e115eb83466946b/firmware/Core/Src/telecommands/telecommand_executor.c#L53) to `... for (uint32_t i = 0; i <= TCMD_timestamp_sent_head && parsed_tcmd->timestamp_sent != 0; i++) ...` Would that be okay for now?

DeflateAwning, 2024-12-13:
Ah fair enough, that's a good change. Surprised no one else has observed that. The check should be bypassed when tssent=0 (although probably not by a condition in the for loop; that'd be a very abnormal style). 

Thanks!

DeflateAwning, 2025-02-12:
I think this was resolved in https://github.com/CalgaryToSpace/CTS-SAT-1-OBC-Firmware/pull/263, right? Can this be closed now?

DeflateAwning, 2025-02-12:
Confirmed that's true ^ :)

closed 2025-02-12 by DeflateAwning