CalgaryToSpace/CTS-SAT-1-OBC-Firmware #268: Telecommand for Filesystem: Delete directory

DeflateAwning, 2025-02-12:
Not sure if you can delete a whole directory, or if you need to recursively delete the files, then delete the directory.

NuclearTea, 2025-05-02:
@DeflateAwning Doing this in 1 shot can trigger the independent watchdog, this may need to create a queue of files that need to be deleted via a FreeRTOS thread? Thoughts?

DeflateAwning, 2025-05-02:
Prefer one-shot in a telecommand. Assume that we'd just format the filesystem if things get truly out-of-hand.

Recursive implementation is okay. While generally not prefered for embedded systems, the max recursion depth is only as deep as the nested folders go. We can assume the folders won't be nested that deep (maybe 3-4 deep only); thus, recursion is acceptable.

NuclearTea, 2025-05-08:
As I am going on vacation, someone else should take this issue.