CalgaryToSpace/CTS-SAT-1-OBC-Firmware #285: GPS: log response into memory

christopherkinyua, 2025-03-12:
Relates to issue #16 

The GPS's response is to be stored in the memory module.

**Initial Concept**

- In the root directory, create a folder called "GPS Logs", where all the log responses go.  
- Within this folder, create files for each response and name the file as the timestamp at which the log is received.
- Incorporate write LFS command into telecommand `TCMDEXEC_gps_send_cmd_ascii` to store the response obtained from the gps.

Areas to consider
- Add checks for the "GPS Logs" directory, for the case it has not been made ( Creation of the directory should happen at start up ) and create the directory.

Deliverable
- Call `TCMDEXEC_gps_send_cmd_ascii` and read the file to ensure it has been logged into the file system

NuclearTea, 2025-05-04:
@DeflateAwning Unsure if this is necessary anymore considering we have `@resp_fname` tag now. Thoughts?

DeflateAwning, 2025-05-12:
Closing this as completed due to the awesome `@resp_fname` telecommand suffix tag.

closed 2025-05-12 by DeflateAwning