CalgaryToSpace/CTS-SAT-1-OBC-Firmware #290: LFS: Write logs to log file in LFS

DeflateAwning, 2025-03-15:
Could setup a periodic job that copies from the in-memory circular buffer to LFS if LFS is too slow.

Otherwise, just need to set up the write on every `LOG_message` call to write to LFS.

2BlackCats, 2025-04-04:
[image]

2BlackCats, 2025-04-04:
@DeflateAwning Doesn't this code already log messages to LFS if sink is enabled?

DeflateAwning, 2025-04-12:
Oh yeah, it sorta does look like that. 

I think we need to revamp it a bit though to keep the log file open, and to rotate log files every hour.

2BlackCats, 2025-04-25:
Unclear what you mean about  keeping the log file open, and to rotate log files every hour.

2BlackCats, 2025-05-05:
Don't have a board or memory module if someone wants to take this issue.

DeflateAwning, 2025-06-01:
Just tested this, and it's looking pretty dire. Enabling the following line fully bricks the satellite after a few log messages go by, as it triggers the watchdog. Definitely need to close log files (flush) less often. 

```
{LOG_SINK_FILE, "log files", LOG_SINK_OFF, LOG_SEVERITY_MASK_DEFAULT}, // Changing to LOG_SINK_ON bricks satellite.
```