CalgaryToSpace/CTS-SAT-1-OBC-Firmware #297: Telecommand: Get all thermal info

parker-research, 2025-03-20:
Create a telecommand which bundles a few related pieces of data.

Bundle this data into a struct, then create a struct to json function, then create a telecommand which sends this. 

### Data:
* OBC temperature
* Antenna temperature
* Power generation per solar panel
* Battery percent
* Battery heater duty cycle (or similar; can't recall the field to use)
* [maybe] MPI temperature (maybe) - needs a function for it

DeflateAwning, 2025-03-20:
Related to #204

kev9268, 2025-03-21:
@DeflateAwning 
Do we want the temperature values to be the raw data or converted to integer (Celsius) to be stored inside of the struct?

kev9268, 2025-03-21:
Values for struct and output are in centiCelcius

kev9268, 2025-03-28:
Changing battery heater cycle field to get battery heat pack and another field to check battery heater status

closed 2026-03-13 by DeflateAwning