CalgaryToSpace/CTS-SAT-1-OBC-Firmware #299: Bug: Warnings about stack overflows

DeflateAwning, 2025-03-20:
Fix the bug where this warning is emmitted frequently, seemingly-incorrectly.

If we can't solve what's causing it, then we should just get rid of the thing that logs that.

```
0000000000000+0000017170_N [OBC:WARNING]: Warning: Task 'TASK_DEBUG_print_heartbeat' approached a stack overflow. Worst remaining stack size was: 0 bytes.
0000000000000+0000017190_N [OBC:WARNING]: Warning: Task 'TASK_service_eps_watchdog' approached a stack overflow. Worst remaining stack size was: 0 bytes.
0000000000000+0000017204_N [OBC:WARNING]: Warning: Task 'TASK_time_sync' approached a stack overflow. Worst remaining stack size was: 0 bytes.
```

2BlackCats, 2025-04-04:
@DeflateAwning As the warning implies the stack appears to be too small. We can either remove the warning message or increase the stack size. What would you prefer?

2BlackCats, 2025-04-25:
uxTaskGetStackHighWaterMark is being set to 0

2BlackCats, 2025-04-25:
` // Get the highstack watermark
            const uint32_t task_min_bytes_remaining = uxTaskGetStackHighWaterMark(task_handle) * 4;

            if (task_min_bytes_remaining < FREERTOS_task_handles_array[task_num].lowest_stack_bytes_remaining) {
                // If this is the new "lowest free space", update that value.
                FREERTOS_task_handles_array[task_num].lowest_stack_bytes_remaining = task_min_bytes_remaining;

                // Determine the threshold of the task
                const uint32_t task_threshold_bytes = (
                    FREERTOS_task_handles_array[task_num].task_attribute->stack_size
                    * CONFIG_freertos_min_remaining_stack_percent
                    / 100
                );
                
                // If this new "lowest free space" is below the threshold, warn the user.
                if (task_min_bytes_remaining < task_threshold_bytes) {
                    LOG_message(
                        LOG_SYSTEM_OBC, LOG_SEVERITY_WARNING, LOG_SINK_ALL,
                        "Warning: Task '%s' approached a stack overflow. Worst remaining stack size was: %lu bytes.",
                        pcTaskGetName(task_handle),
                        task_min_bytes_remaining
                    );
                }`

2BlackCats, 2025-04-25:
https://www.freertos.org/Documentation/02-Kernel/04-API-references/03-Task-utilities/04-uxTaskGetStackHighWaterMark

2BlackCats, 2025-04-25:
Like it could be possible the values are close to zero

2BlackCats, 2025-04-25:
@DeflateAwning