CalgaryToSpace/CTS-SAT-1-OBC-Firmware #367: Bug: LFS read hex command triggers watchdog for large files

DeemDeem52, 2025-04-30:
Documenting this so everyone is aware of it (though I vaguely recall that this one will be removed or modified before flight).

For sufficiently large files (I tested this with a 100 kB file from the ADCS CubeComputer SD card which was placed into the filesystem), calling the LittleFS reading command will cause the files to be continuously read, opened, and closed until for longer than the permitted 15 seconds from the watchdog.

To resolve this, either the function reading must be truncated or it must pet the watchdog periodically.

NuclearTea, 2025-04-30:
The proper solution for this, as discussed with @DeflateAwning is that every `read` function for LFS must have an offset value from where the reading starts. @Saksham-P @KaleF07 @destinsaba Thoughts on this?

KaleF07, 2025-04-30:
That solution makes sense. From my understanding, that would mean the function will start from a given offset and read a certain amount as allowable by the watchdog?

Saksham-P, 2025-04-30:
The solution would solve the issue of reaching the watchdog limit, and just to make sure I understand, would the below things be done to use the function.
- The `read` function for LFS would read a certain amount of data in bytes permitted by the watchdog. 
  - Would we want to add an argument/parameter for the amount of bytes that are read to have more control?
- The user would track and use the correct offset to use for the `read` function to be able to read a whole file?
- The read function already returns the number of bytes read on a successful function call, would we want to make that value be printed out as well for the user?

DeemDeem52, 2025-04-30:
One other thing which I might add... I notice that the function repeatedly opens and closes the file. Might there be a way to keep it open while it reads so that more bytes can be read without triggering the watchdog?

Saksham-P, 2025-04-30:
> I notice that the function repeatedly opens and closes the file

This happens due to the implementation of the telecommand rather than the implementation of the LFS function. The telecommand `TCMDEXEC_fs_read_text_file` uses a while loop to call `LFS_read_file` until all bytes of the file have been read.

NuclearTea, 2025-05-01:
> * Would we want to add an argument/parameter for the amount of bytes that are read to have more control?
>     * The user would track and use the correct offset to use for the `read` function to be [...]

1. This is one approach @DeflateAwning, would love your input here. Alternatively, we could hard code it to be 512 bytes (or something like that)
2. Yes.
3. Yes. This is because its possible that the amount that they read will be less than the allowed. For example, I have a file that contains:
```
Hello World!
```
however, I try to read 100 bytes from this file. I will only read 13 bytes as that is the length of the entire file, so it would be helpful to have the number of bytes read regardless.

DeflateAwning, 2025-05-02:
For all these filesystem reads (hex, str, maybe others), can we please change them to have the following arguments roughly:

1. File path
2. Start offset ("offset")
3. Max bytes to link ("length"), or 0 for automatic selection of `min("maximum allowable", "file size")`

Let "maximum allowable" be just under the max that will fit in `response_output_buffer_len`.

Please pass the output through that buffer instead of through log/debug UART print str. 

If this feels like it's not quite a good enough solution; don't worry - These are mostly "backup" telecommands for downlinking files. The nominal way to downlink [large] files is described in #342

PR #468 (2025-06-01): fix: Add chunking arguments/limits to `fs_read_` telecommands
Fixes #367
closed 2025-06-01 by DeflateAwning