CalgaryToSpace/CTS-SAT-1-OBC-Firmware #372: telecommand "mpi_send_command_hex" times out after first try

JohnathanBurchill, 2025-05-02:
MPI responds with correct response bytes on first use of the telecommand. Subsequent calls time out. Resetting the MPI state by powering down the MPI 12 V and 5 V lines does not reset the issue. Power cycling the OBC resets the behaviour.

DeflateAwning, 2025-05-03:
Potentially fixed in `da/mpi-commands-update` branch in commit fc5807f

JohnathanBurchill, 2025-05-03:
Commit a6faa759 sets the rx buffer Len when breaking out of the `while(1)` loop when the receive buffer is full.

DeflateAwning, 2026-01-11:
I think this is fixed. Will confirm in testing this week hopefully.

DeflateAwning, 2026-02-02:
@meagan-davies and Aarti tested this in the lab in 2026-01. Will leave it to them to determine whether this issue can be closed now!