CalgaryToSpace/CTS-SAT-1-OBC-Firmware #420: Start-of-mission/post-ejection/bootup sequence and operation state machine

DeflateAwning, 2025-05-16:
When the satellite boots up (e.g., on a workbench during integration into the launch vehicle, upon deployment from the launch vehicle, rebooted in space, after a crash/watchdog reset in space), it will enter the `CTS1_OPERATION_MODE_BOOTED_AND_WAITING` state. This issue describes a state machine of how the satellite deploys its antennas safely, with fallbacks if single-system failures occur.

## TL;DR

* When the satellite boots with the RBF tag installed, the antennas won't deploy. The radio will never transmit. All other systems are operational.
* When the satellite boots with the RBF tag removed (i.e., floating through space), the satellite will wait 30 minutes, then deploy its antennas, then move onto its nominal state with radio transmissions enabled. 
* When the satellite reboots in space, the RBF tag will be removed (obviously). The satellite has several bypass checks to avoid waiting 30 minutes 
    * These checks are: Are the antennas already deployed? Are uplinks happening? Has a "deployment successful" flag been written to the filesystem?
* After we've determined the satellite has deployed successfully, create a file containing "1" named "/bypass_deployment_and_enable_radio.txt" in the filesystem.

## Precautions
1. **Do not uplink to the satellite with its antennas stowed.** If the satellite receives a radio uplink in any state, it will assume its antennas are deployed and will enable radio operation.
2. Only change the RBF tag state when the satellite is powered off.
    * Reason 1: State transitions while powered on are untested/unvalidated.
    * Reason 2: Timed events are based on system uptime, not duration since RBF state change.
3. Obviously, when the RBF tag is removed, the satellite MUST NOT sit on a bench with its deployment switches deployed for more than 25 minutes! It will deploy its antennas. 
4. Do not create the `/bypass_deployment_and_enable_radio.txt` file until floating through space (antennas deployed, etc.). Even then, it's only helpful if ANTS fails (i.e., can't use it to check that antennas are deployed) and/or 

### States

* Global variable `enum CTS1_operation_mode_enum_t CTS1_operation_mode`:

```c
typedef enum {
    CTS1_OPERATION_MODE_BOOTED_AND_WAITING, // Entry point. Booted, and awaiting a trigger to deployment.
    CTS1_OPERATION_MODE_DEPLOYING, // Intermediate state. Very short-lived nominally. Deploy deployable antennas here.
    CTS1_OPERATION_MODE_NOMINAL_WITH_RADIO_TX,
    CTS1_OPERATION_MODE_NOMINAL_WITHOUT_RADIO_TX
} CTS1_operation_mode_enum_t
```

## State Transitions

### Conditions checked first in every state

* Condition 1: If any uplink has been received - `if (AX100_uptime_at_last_received_kiss_tcmd_ms > 1000)` -> NOMINAL_WITH_RADIO_TX
* Condition 2: If `(RBF_STATE == BENCH)` -> NOMINAL_WITHOUT_RADIO_TX
* Condition 3: If `/bypass_deployment_and_enable_radio.txt` file exists in filesystem -> NOMINAL_WITH_RADIO_TX

### BOOTED_AND_WAITING to <other state>

* Condition 3: If `(RBF_STATE == FLYING) && (uptime > 30 minutes)` -> DEPLOYING
    * Reason: Nominal post-ejection transition
* Condition 4: If `(RBF_STATE == BENCH)` -> WITHOUT_RADIO_TX
    * Reason: On bench testing, bypass the countdown, but disable the radio.
* Condition 5: If all four antennas have been deployed - check using `ANT_` commands -> NOMINAL_WITH_RADIO_TX
     * Reason: On reboot after day 1 of mission, bypass the waiting stage.
* Condition 6: If `/bypass_deployment_and_enable_radio.txt` file exists -> NOMINAL_WITH_RADIO_TX
    * Reason: On reboot after day 1 of mission, bypass the waiting stage. This serves as an override if 

### DEPLOYING

* Condition 7: If deployment successful -> NOMINAL_WITH_RADIO_TX
* Condition 8: If `uptime > 4 hours` -> NOMINAL_WITH_RADIO_TX
    * Reason: If deployment goes wrong for 3.5 hours, assume the issue is that the "is deployed" sensors failed.

### NOMINAL_WITH_RADIO_TX

No additional exit conditions. Only the all-state checks at the start.

### NOMINAL_WITHOUT_RADIO_TX

No additional exit conditions. Only the all-state checks at the start.

## State Behaviours

### BOOTED_AND_WAITING

* No radio TX.
* All systems are operational.
* Telecommands sent over debug UART are still processed and executed. 
* From boot until 25 minutes uptime, external LED pulses 40ms per 1000ms. 
* From 25 minutes uptime to 30 minutes uptime, external LED blinks 40ms per 333ms (warning indicator that the satellite is about to deploy its antenna).

### DEPLOYING

* No radio TX.
* All systems are operational.
* Telecommands sent over debug UART are still processed and executed. 
* Antenna deployment signal is sent to deployment antenna. 

### NOMINAL_WITH_RADIO_TX

* Radio TX enabled.
* All systems are operational.

### NOMINAL_WITHOUT_RADIO_TX

* Radio TX disabled.
* All systems are operational.

## Design inputs: Hard Requirements/Existing Constraints

1. When the RBF tag is removed, the satellite must not beacon nor deploy anything for 30 minutes upon deployment (i.e., upon initial power-on).
2. When the RBF tag is present, the satellite must NEVER deploy anything, and must NOT beacon.
3. The procedure must function no matter the bootup environment.

## Minor Implementation Details

1. To determine whether the RBF tag is present, read that pin twice separated by 300ms (in case noise somehow causes a single unreliable reading). 
2. Do not check the contents of the `/bypass_deployment_and_enable_radio.txt` file. Simply check for its presence.
3. To deploy the antenna, use the `ANT_CMD_arm_antenna_system()` function, and then the `ANT_CMD_start_automated_sequential_deployment()` functions.

This issue replaces #27 and #148, both of which don't quite consider all usages and requirements. Those issues fail to utilize the RBF tag appropriately.

JohnathanBurchill, 2025-05-26:
Let me know if that picture is consistent with the state machine (aside from the addition of a 'booting' state. Whether that is needed in practice is not clear to me. I put it in to keep the transition rules orderly).

DeflateAwning, 2025-05-28:
### Re: Diagram

Thanks for drawing that! Very nice to see it written out like that.

1. I really like the way you used `AND` and `OR` operators to make it clear to read. There's a few place missing those operators though. In an effort to validate interpretatability of the plan, I'll leave it as an "exercise to the reader" to place those.
2. The BOOTING state doesn't exist in the implementation. The idea is more-so that these are states in which the next-state logic has a timeslice of CPU execution (i.e., the next-state code is being executed). As such, the period when the satellite is powering on, initializing peripherals, etc. (presumably what's represented with "BOOTING") is intentionally not considered. I think the diagram might be better-suited without it in order to align to the spec and code; at the very least, it shouldn't be a state with several transitions out of it. The diagram will also get at least a bit simpler without it.
3. Discrepancy: `BOOTED_AND_WAITING` TO `NOMINAL_WITHOUT_RADIO_TX` occurs independent of uptime.
4. The loopback conditions in the two NOMINAL modes, for completeness, would need `AND NOT <condition to transfer out of that state>`

### Re: Procedure vs. Firmware Spec

Great observation on the procedure vs. implementation spec; this was actually 100% intentional. The procedure was written intentionally stricter than the spec/firmware is implemented, in order to create clear and concise instructions, and to create a far-simpler mental model of the satellite's state.

The firmware is implemented with a `NOMINAL_WITH_RADIO_TX -> NOMINAL_WITHOUT_RADIO_TX` (based on RBF state) in case a sleepy/inexperienced/rushed/stressed person forgets to reboot the satellite after re-installing the RBF tag upon accidental removal (and even inadvertent deployment). Having that condition checked in all states also simplifies the code (e.g., for auditability). 

The brief procedure [which aims to serve as inspiration to the actual integration procedure] "requires" rebooting after such state changes, which guarantees a known and easy-to-follow path through the norminal flow chart during integration and testing. It also resets the uptime counter, which aids in the repeatability aspect.

closed 2025-05-28 by DeflateAwning