CalgaryToSpace/CTS-SAT-1-OBC-Firmware #488: Configuration Variable: Maximum MPI active duration

DeflateAwning, 2025-06-02:
Add a configuration variable for maximum duration (in seconds) that the MPI can be in the "active recording data" mode. 

Then, if it exceeds that duration, turn it off in the MPI task.

This is important because the MPI is very power-hungry (about 3W), and we want to make sure it powers off in case something goes wrong during active mode and the scheduled telecommand to turn it off fails to execute.

Freddydeddydoo, 2025-06-04:
Is there any other information I need to know before starting this assignment?

Freddydeddydoo, 2025-06-04:
This is what I will do:

- Create telecommand to configure maximum duration,
- Create the config variable itself.

Freddydeddydoo, 2025-06-18:
Should I use a timer interrupt or simply hal_getTick?

DeflateAwning, 2025-06-18:
`HAL_GetTick()` - returns the system uptime in ms.